ghdl/ghdl #2070: Fuzz testing GHDL syntax checker

sudden6, 2022-05-31:
**Description**
I tried to run fuzz testing on the VHDL parser of GHDL. After some weekends of tinkering, I managed to build GHDL with the instrumentation required to use [AFLplusplus](https://github.com/AFLplusplus/AFLplusplus).

I chose `ghdl -s --std=08 <fuzzing input>` as my target for fuzzing, because it's one of the simplest commands that trigger the parser. Additionally, as far as I can tell this uses the exact same parser as for simulation, making it possible to extend this work for some deeper testing.

After 24h of fuzzing on one core of my laptop and some result cleanup to perform basic crash de-duplication the attached set of crashing input files was produced:
[crashes-dedup.zip](https://github.com/ghdl/ghdl/files/8809842/crashes-dedup.zip)

Each of these files crashes GHDL. There might still be duplicates as I removed only duplicates with the exact same coverage path or the exact same content.

**Expected behaviour**
I expect GHDL to handle arbitrary input without triggering a `GHDL Bug occurred` message. Error messages about invalid or unsupported syntax are ok.

**How to reproduce?**
I'll provide a docker file and the necessary scripts to reproduce these results. If there's interest we can also try to integrate this into CI.

tgingold, 2022-06-01:
Interesting.  It has found some overflow issues.
I have fixed the crashes, and added the tests in the testsuite.

umarcor, 2022-06-01:
> I'll provide a docker file and the necessary scripts to reproduce these results. If there's interest we can also try to integrate this into CI.

@sudden6 please upload those to a fork/branch of yours and let us know where it is. Since Linux CI jobs are executed using containers, the dockerfile might need to be added in ghdl/docker, to then consume it here.

commit 15ca98bf (2022-06-01): testsuite/gna: add tests for #2070

Close #2070