ghdl/ghdl #2074: synth_dyadic_operation: unhandled IIR_PREDEFINED_TF_ARRAY_AND with bit_vectors and bitwise operations

robinsonb5, 2022-06-05:
**Description**
Bitwise operations such as q <= a and b; fail in synthesis if q, a and b are bit_vectors - but work fine in simulation.

**Expected behaviour**
I think bitwise operations are legal on bit_vectors, so should be accepted?

**How to reproduce?**
Here's a minimal example:
```
library ieee;
use ieee.std_logic_1164.all;

entity bitvec is
port (
	clk : in std_logic;
	d : in bit_vector(7 downto 0);
	q : out bit_vector(7 downto 0)
);
end entity;

architecture rtl of bitvec is
	constant a : bit_vector(7 downto 0) := X"5a";
begin
	q <= d and a;
end architecture;
```
ghdl -a bitvec.vhd
ghdl --synth bitvec
produces the message "bitvec.vhd:15:16:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_TF_ARRAY_AND"

- OS: Linux Mint 20.3
- Origin: Built from sources: 9a001446dd281c47a8321ce475bcf1e79dc1b859

tgingold, 2022-06-05:
I will fix that.  Support of bit/bit_vector in synthesis is not complete.  I haven't found many designs using them.
(In particular I fear there is currently no support for numeric_bit, although it's not difficult to add).

tgingold, 2022-06-05:
Should be fixed.  Thank you for the issue.

commit 137641db (2022-06-05): synth-vhdl_oper: handle more bit_vector operations.  Fix #2074
closed 2022-06-05 by tgingold