ghdl/ghdl #2075: testsuite/gna/issue2070/crash17.vhdl fails for LLVM and mcode backends

Xiretza, 2022-06-05:
**Description**
For some reason, the testsuite for `ghdl-llvm-git` and `ghdl-mcode-git` AUR packages fails on the fuzzer outputs from #2070. GCC backend is fine, and CI tests are also fine, so I'm not entirely sure what's going on. Here's the crash output:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r304.g9a001446d) [Dunoon edition]
Compiled with GNAT Version: 12.1.0
Target: x86_64-pc-linux-gnu
/build/ghdl-mcode-git/src/ghdl/testsuite/gna/issue2070/
Command line:
/build/ghdl-mcode-git/src/ghdl/ghdl_mcode -a --expect-failure --std=08 crash17.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : vhdl-nodes.adb:910 range check failed
Call stack traceback locations:
0x559bf9d25d56 0x559bf9d32e4c 0x559bf9db3cad 0x559bf9db0a39 0x559bf9db2841 0x559bf9db7394 0x559bf9db7fce 0x559bf9e0e924 0x559bf9e0e9b1 0x559bf9e5e8fa 0x559bf9dc1608 0x559bf9ee6f79 0x559bf9c2e01d 0x7f5e72db228e 0x7f5e72db2348 0x559bf9c2e093 0xfffffffffffffffe
******************************************************************
```

**Expected behaviour**
The testsuite succeeds.

**How to reproduce?**

```
export ADA_FLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -fstack-clash-protection -fcf-protection -g -fvar-tracking-assignments -ffile-prefix-map=/build/ghdl-mcode-git/src=/usr/src/debug"

./configure \
	--prefix=/usr/ \
	--enable-libghdl \
	--enable-synth

make

make install.vpi.local
make test
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - Built from sources: 9a001446d

tgingold, 2022-06-06:
I was not able to reproduce it.  Can you try under gdb and disp a backtrace at the crash point ?

Usually:
gdb /build/ghdl-mcode-git/src/ghdl/ghdl_mcode -a --expect-failure --std=08 crash17.vhdl
(gdb) catch exception
(gdb) run
...
(gdb) bt

Xiretza, 2022-06-07:
I seems to have been fixed at some point between 9a00144 and 0464fc0.

closed 2022-06-07 by Xiretza