ghdl/ghdl #2076: Fuzz testing GHDL syntax checker round 2

sudden6, 2022-06-05:
Unfortunately I can't reopen #2070 so I decided to create a new issue, I hope that's ok.

I have continued to run fuzz testing and to try and build a meaningful corpus of input data and found some more crashing inputs, which I attached. [crashes-2022-06-05.zip](https://github.com/ghdl/ghdl/files/8840706/crashes-2022-06-05.zip)

As promised earlier, I also put the workflow into scripts that might be useful for a possible CI integration, or at least simplified manual testing. It is certainly a proof of concept and not as simple to use as I had hoped. I attached another zip file with that. 
[ghdl_afl_fuzzing.zip](https://github.com/ghdl/ghdl/files/8840739/ghdl_afl_fuzzing.zip)

A short overview on how these fuzzing scripts work:

## Dockerfile

The docker file builds GHDL with the compiler wrapper provided by AFL++ (https://github.com/AFLplusplus/AFLplusplus).
  - Since GHDL is written in Ada, we must use the GCC plugin mode of AFL++. Ada, GCC and AFL++ is an exotic combination I have never tried so far and there's not much to find on the internet either, but it seems to work.
  - A patch is applied to `bug.adb` which disables handling of `SIGABRT` and SIGSEGV in the Ada runtime, because AFL++ overrides the signals itself to detect crashes. This is a very important point to fuzz Ada programs. It also modifies the bug handler to trigger `SIGABRT` when something unexpected happens. Triggers to send `SIGABRT` can theoretically placed anywhere to investigate interesting behaviours or to minimize existing crashing testcases using the `afl-tmin` tool.

## fuzz.sh

The ´fuzz.sh` script basically implements the flow from [AFL++ docs] (https://aflplus.plus/docs/) and is used interactively from the docker container.

- The `fuzz` command starts the main fuzzer, it can be stopped with [Ctrl+C] at any time. It takes the files from `/work/corpus/`
- The `corpus_prune` takes the corpus after a fuzzing run and performs the following steps:
  - Rename the files to their SHA256 hash to make the names shorter and avoid duplicates
  - Run `afl-cmin` over the corpus, this removes files from the corpus that don't bring in any new coverage
  - Run `afl-tmin` on each file, to erase blocks from the file that don't trigger new coverage
  - Rename the files to their hashes again and save the results in `corpus_new`
  - On the next run, the files in `/work/corpus/` should be replaced with the files in `corpus_new`, because they are the smallest and fewest files which still have the same coverage.

- The `crash_prune` command does the same, but for crashes found during fuzzing. This is useful to reduce the set of crashes that must be debugged.

## run.sh

This just builds and launches the docker container with the correct mounts to persistently store the corpus.

I'd be happy to hear your feedback and if we should work together to integrate this into the testing scheme of GHDL.

tgingold, 2022-06-06:
Did you use particular options to run the tests ?  I can find only 9 crashes within these > 170 files.

sudden6, 2022-06-06:
There may be duplicates, where different inputs (e.g. upper case / lower case) yields the same crash, but the code execution until it crashes is different enough that AFL++ recognizes it.

The exact options I used are in the `fuzz.sh` script.

tgingold, 2022-06-06:
Ah ok, thanks.
Indeed, there was another issue to fix.

closed 2022-06-14 by tgingold