ghdl/ghdl #2077: synth: crash in memory inference for arrays of memories

Xiretza, 2022-06-06:
**Description**
When declaring an array of memories, ghdl recognizes the entire thing as a single memory and subsequently crashes.

**Expected behaviour**
GHDL infers one memory per array element (24 of them in this case, each with one sync read port and one sync write port), and doesn't crash.

**How to reproduce?**

```vhd :file: ent.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
	generic (
		WIDTH : positive := 32;
		DEPTH : positive := 256;

		WAYS : positive := 24
	);
	port (
		clk: in std_logic;

		write_enable: in std_logic;
		active_way: in natural range 0 to WAYS-1;
		write_address: in natural range 0 to DEPTH-1;
		input: in std_logic_vector(WIDTH-1 downto 0);

		read_address: in natural range 0 to DEPTH-1;
		outputs: out std_logic_vector(WAYS-1 downto 0)
	);
end entity;

architecture a of ent is
begin
	process(clk)
		type memory_t is array(0 to DEPTH-1) of std_logic_vector(WIDTH-1 downto 0);
		type memories_t is array(0 to WAYS-1) of memory_t;

		variable memories : memories_t;
	begin
		if rising_edge(clk) then
			for i in 0 to WAYS-1 loop
				outputs(i) <= and memories(i)(read_address);
			end loop;

			if write_enable then
				memories(active_way)(write_address) := input;
			end if;
		end if;
	end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
$ ghdl analyze --std=08 ent.vhdl
$ ghdl synth --std=08 ent
ent.vhdl:31:26:note: found RAM "n1.memories", width: 32 bits, depth: 6144
                variable memories : memories_t;
                         ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r260.g45303bcb5) [Dunoon edition]
Compiled with GNAT Version: 12.1.0
Target: x86_64-pc-linux-gnu
/tmp/
Command line:
ghdl synth --std=08 ent
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists-memories.adb:925
Call stack traceback locations:
0x5e80a1 0x451e43 0x52939a 0x546df9 0x5292c7 0x5beb63 0x50bfec 0x40aacd 0x7f04830b928e 0x7f04830b9348 0x40ab43 0xfffffffffffffffe
******************************************************************
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - Package manager: `ghdl-gcc-git 3.0.0dev.r7252.g45303bcb5-1`

tgingold, 2022-06-06:
I saw you opened several issues about memory.
I will investigate them but later.  I know the algorithm is very fragile.

Xiretza, 2022-06-06:
Yeah, no problem, I have workarounds for all of them - just thought I'd leave them here so they can be fixed eventually.

dalex78, 2026-08-20:
# Issue #2077 — Crash inferring memory for an array of memories

Analysis by an AI agent.

## High-level summary

Declaring an array of memories (`type memories_t is array(0 to WAYS-1) of memory_t;`, each element itself a RAM array) crashed `ghdl --synth` with an internal `ASSERTION_ERROR` at `netlists-memories.adb:925`, instead of either inferring one memory per array element (as the reporter expected) or failing gracefully.

## Investigation / current status: already fixed and already tested upstream

Before writing a new reproducer, checked whether `master` already had a test for this issue — it does: `testsuite/synth/issue2077/` already exists, with 5 variant entities (`ent1`..`ent6`, `ent1` being essentially the report's exact repro) and a `testsuite.sh` that passes in full on this build:

```
Synthesis of ent1.vhdl -e
note: memory "n1.memories" uses different depth
  ...
[ent2/ent3/ent5/ent6: "found RAM", synthesized, netlist re-analyzed and
 simulated against a testbench, all passing]
Test successful
```

`ent1` (the report's exact case, with a dynamic `active_way` index selecting which memory to write) is explicitly commented `-- Not yet handled.` in the upstream test and only gets a `synth_only` check (no crash, no further correctness verification) — matching what independent manual reproduction of the original report showed here: no crash, but GHDL falls back to a single flattened bit-vector netlist instead of inferring `WAYS` separate memories. `ent2`/`ent3`/`ent5`/`ent6` (other array-of-memories shapes, with a statically-known or more constrained write-way) are fully supported and verified end-to-end (synthesized, re-analyzed, and simulated against a testbench).

So: the crash this issue reports is fixed, and upstream has already both fixed and tested the fixable subset of this pattern. No source change or new test was needed — there is nothing further to add here without implementing genuinely new memory-inference support for the dynamic-index case, which is out of scope for a bug-fix pass (and is arguably still an open feature request under this same issue, not a bug, since `ent1` already degrades gracefully rather than crashing).

## What the fix does

No source change and no new test file in this commit — `master` already has a complete, passing regression test for this issue at `testsuite/synth/issue2077/`.

dalex78, 2026-08-20:
Recommend to close that ticket as already done @tgingold.

dalex78, 2026-08-20:
# Issue #2077 — Crash inferring memory for an array of memories

Analysis by an AI agent.

## High-level summary

Declaring an array of memories (`type memories_t is array(0 to WAYS-1) of memory_t;`, each element itself a RAM array) crashed `ghdl --synth` with an internal `ASSERTION_ERROR` at `netlists-memories.adb:925`, instead of either inferring one memory per array element (as the reporter expected) or failing gracefully.

## Current status: the crash is fixed, the inference is not implemented

The crash is gone on current `master`, and upstream already has a test for this issue at `testsuite/synth/issue2077/`. But that test does not claim the reported case works — it says the opposite:

```sh
# Not yet handled.
synth_only ent1

for t in ent2 ent3 ent5 ent6; do
    synth_tb $t
done
```

`ent1` is essentially the report's own design, with a dynamic `active_way` index selecting which memory to write. It gets `synth_only`: GHDL must not crash, and nothing further is checked. Reproducing it by hand agrees — no crash, but GHDL falls back to a single flattened bit-vector netlist instead of inferring `WAYS` separate memories. The shapes that *are* fully supported (`ent2`, `ent3`, `ent5`, `ent6` — a statically-known or more constrained write way) are synthesized, re-analyzed and simulated against a testbench.

So the outcome splits in two:

- **Fixed:** the internal error the issue reports. GHDL now degrades gracefully.
- **Not implemented:** inferring one memory per element when the write index is dynamic, which is what the reporter was after.

Nothing in the thread claims otherwise. tgingold's only comment is *"I saw you opened several issues about memory. I will investigate them but later. I know the algorithm is very fragile."*, and the reporter replied *"I have workarounds for all of them - just thought I'd leave them here so they can be fixed eventually."* — i.e. left open deliberately, awaiting a fix rather than a workaround.

## Recommendation

Do not close this as done. The crash is fixed and already covered upstream, so there is no source change or new test to add here, but the feature the issue asks for is still missing and upstream's own test records that. It is better tracked as an open enhancement for dynamic-index array-of-memories inference.

## What this commit does

No source change and no new test — `master` already covers what has actually been fixed, at `testsuite/synth/issue2077/`.