ghdl/ghdl #2079: synth: crash on async memory write port

Xiretza, 2022-06-06:
**Description**
When a memory with an async write port is attempted to be inferred, ghdl crashes instead of producing an error message.

**Expected behaviour**
GHDL produces an error message explaining the problem.

**How to reproduce?**
```vhd :file: ent.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
	generic (
		WIDTH : positive := 32;
		DEPTH : positive := 256
	);
	port (
		clk: in std_logic;

		write_enable: in std_logic;
		write_address: in natural range 0 to DEPTH-1;
		input: in std_logic_vector(WIDTH-1 downto 0);

		read_address: in natural range 0 to DEPTH-1;
		output: out std_logic_vector(WIDTH-1 downto 0)
	);
end entity;

architecture a of ent is
begin
	proc: process(clk)
		type memory_t is array(0 to DEPTH-1) of std_logic_vector(WIDTH-1 downto 0);
		variable memory : memory_t;
	begin
		output <= memory(read_address);

		if write_enable = '1' then
			memory(write_address) := input;
		end if;
	end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
$ ghdl synth --std=08 ent.vhdl -e
ghdl:note: top entity is "ent"
ent.vhdl:26:26:note: found RAM "proc.memory", width: 32 bits, depth: 256
                variable memory : memory_t;
                         ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r260.g45303bcb5) [Dunoon edition]
Compiled with GNAT Version: 12.1.0
Target: x86_64-pc-linux-gnu
/tmp/
Command line:
ghdl synth --std=08 ent.vhdl -e
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists-memories.adb:2057
Call stack traceback locations:
0x5e80a1 0x451b88 0x52939a 0x546df9 0x5292c7 0x5beb63 0x50bfec 0x40aacd 0x7fc8d81b828e 0x7fc8d81b8348 0x40ab43 0xfffffffffffffffe
******************************************************************
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - Package manager: `ghdl-gcc-git 3.0.0dev.r7252.g45303bcb5-1`

Paebbels, 2022-06-06:
This design is not asynchronous. It's a double data rate memory, because no `if rising_edge` is used, so it's sensitive to both edges. Also write operation is synchronous, because of sensitivity list.

Xiretza, 2022-06-06:
The sensitivity list has no effect on synthesis though, right?

umarcor, 2022-06-08:
@Xiretza, my understanding is that's what several vendors do, but it's not what the LRM says. So, the sensitivity is ignored *de facto* but it should not be.

dalex78, 2026-08-20:
# Issue #2079 — Crash on async memory write port

Analysis by an AI agent.

## High-level summary

A memory written unconditionally inside a `clk`-sensitive process with no `if rising_edge(clk)` guard (i.e. an asynchronous/unclocked write, along with an unclocked read) crashed `ghdl --synth` with an internal `ASSERTION_ERROR` at `netlists-memories.adb:2057`, instead of the clean diagnostic the reporter expected ("GHDL produces an error message explaining the problem").

## Investigation / current status: already fixed on master

Checked `master` didn't already have a test for this issue first (per the process fix noted from #2077) — it didn't. Reproduced the exact repro from the report against current `master`:

```
ent.vhdl:26:26:error: latch infered for net "proc.memory" (use --latches)
                variable memory : memory_t;
```

`--synth` now correctly recognizes this pattern (an unclocked read/write inside a `clk`-sensitive process) as inferring a latch and reports a clean, located error — exactly matching the reporter's requested behavior — instead of crashing. This must have been fixed as part of general improvements to latch-vs-memory inference in `src/synth` since this was reported (originally against a 2022 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue2079/` (the exact repro from the report) whose `testsuite.sh` attempts `--synth`, expects it to fail, and asserts the clean "latch infered" error appears with no `GHDL Bug occurred` crash. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this issue be closed ? (PR has been merged)

closed 2026-08-24 by tgingold