ghdl/ghdl #2082: pyGHDL crash when parsing sensitivity lists containing arrays

eng73, 2022-06-08:
**Description**
pyGHDL crashes when attempting to parse a sensitivity list that includes an index of an array. This was discovered when trying to parse Xillinx module `tri_mode_ethernet_mac_rx_client_fifo.vhd`.

**Expected behaviour**
Support for arrays in sensitivity list would be ideal however something that is recoverable such as raising a NotImplemented error would suffice.

**How to reproduce?**
Using the following as `test.vhd`.

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_entity is port (
  f_clk : in std_logic;
  b_mode_o : out std_logic
);
end test_entity;

architecture RTL of test_entity is 

  type cntl_pipe is array (0 to 2) of std_logic;
  signal test_pipe : cntl_pipe;

begin
  test_process_p : process(test_pipe(1))
  begin
    b_mode_o <= '1';
  end process;

end architecture;
```
and parsing using Python:
```python
from pathlib import Path
from pyGHDL.dom.NonStandard import Document

source_file = Path("test.vhd")
Document(source_file)
```
Creates the following error:
```sh
 raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Identifier
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin:
  - [x] Built from sources: `d2d8d12ab243deafc63b202ad9e6603e6454b2b5`

Paebbels, 2022-06-08:
@tgingold This looks like an internal error raised by libghdl, but not pyGHDL itself, right?

tgingold, 2022-06-08:
The sensitivity list of the DOM is a list of string.  So I suppose this should be moved to a list of nodes.