ghdl/ghdl #2086: raised TYPES.INTERNAL_ERROR : netlists-memories.adb:1419 (unimplemented code path?)

LAK132, 2022-06-09:
I'm hitting 
```ada
      if Negate then
         --  TODO.
         raise Internal_Error;
      end if;
```
in `In_Conjunction`.

Unfortunately I can't create a small example that triggers this behaviour, seems that some level of complexity is required to get here.

**Context**

- OS:
  - Windows 10 (Ubuntu 21.04 WSL)
- Origin:
  - [ ] Built from sources: `634b0991bd2709fc19eec5d0af0f46a52341b70c`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/mnt/d/Repos/mega65-core-toolchain/mega65-core/build/mega65r3/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-memories.adb:1419
******************************************************************
```

tgingold, 2022-06-09:
A non-small example would be nice to investigate.

LAK132, 2022-06-10:
With these files:
https://github.com/MEGA65/mega65-core/blob/6b5d01e61fd8fb7eea110a056321d674f3114e68/src/vhdl/hdmi_tx_encoder.vhdl
https://github.com/MEGA65/mega65-core/blob/6b5d01e61fd8fb7eea110a056321d674f3114e68/src/vhdl/types_pkg.vhdl
https://github.com/MEGA65/mega65-core/blob/6b5d01e61fd8fb7eea110a056321d674f3114e68/src/vhdl/vga_to_hdmi.vhdl (with references to xpm commented out)

Run:
`ghdl --synth types_pkg.vhdl vga_to_hdmi.vhdl hdmi_tx_encoder.vhdl -e vga_to_hdmi`

Gets:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-linux-gnu
/mnt/d/Repos/mega65-core-toolchain/mega65-core/
Command line:
/mnt/d/Repos/mega65-core-toolchain/ghdl/build/bin/ghdl --synth source/core/types_pkg.vhdl source/core/vga_to_hdmi.vhdl source/core/hdmi_tx_encoder.vhdl -e vga_to_hdmi
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-memories.adb:1419
Call stack traceback locations:
0x560e536c46a5 0x560e536c485e 0x560e536c49d0 0x560e536c499b 0x560e536c8a77 0x560e5380c87c 0x560e53846a46 0x560e53846a92 0x560e5380c7e7 0x560e538e6090 0x560e537dfb96 0x560e5399a035 0x560e5357350b 0x7f9e10356563 0x560e5357133c 0xfffffffffffffffe
******************************************************************
```

commit ea07fa6b (2022-06-11): netlists-memories: handle negation for In_Conjunction.  Fix #2086
closed 2022-06-11 by tgingold