ghdl/ghdl #2093: synth: Verilog output should escape Verilog keywords

antonblanchard, 2022-06-13:
The Microwatt `bit_counter` module has a signal called `edge`. This is a Verilog keyword, and Verilator complains:

```
%Error: microwatt.v:4086:21: syntax error, unexpected edge, expecting TYPE-IDENTIFIER
```

We should detect all Verilog keywords, and escape them (ie "`\edge `")

Xiretza, 2022-06-13:
Duplicate of #2054 related to #1154.

antonblanchard, 2022-06-13:
Thanks @Xiretza, I should have checked. Closing.

antonblanchard, 2022-06-15:
Reopening because I don't think the fix for #2054 resolves the Microwatt issue. A cut down test case:

```
library ieee;
use ieee.std_logic_1164.all;

entity testcase is
    port (
        clk : in std_logic;
        i   : in std_ulogic_vector(63 downto 0);
        o   : out std_ulogic_vector(63 downto 0)
        );
end entity testcase;

architecture behaviour of testcase is
    signal edge : std_ulogic_vector(63 downto 0);
begin
    testcase_0: process(clk)
    begin
        if rising_edge(clk) then
            edge <= i;
            o <= edge;
        end if;
    end process;
end behaviour;
```

Creates a verilog wire called edge:

```
# ghdl --synth --out=verilog --std=08 testcase.vhdl -e testcase > testcase.v

# verilator --lint-only testcase.v
%Error: testcase.v:5:15: syntax error, unexpected edge, expecting IDENTIFIER or do or final or randomize
    5 |   wire [63:0] edge;
      |               ^~~~
%Error: testcase.v:10:10: syntax error, unexpected edge
   10 |   assign edge = n7_q;
      |          ^~~~
%Error: testcase.v:13:13: syntax error, unexpected edge, expecting TYPE-IDENTIFIER
   13 |     n6_q <= edge;
      |             ^~~~
%Error: Exiting due to 3 error(s)
```

commit 18613eba (2022-06-15): netlists-rename: handle handle signal instances.  Fix #2093
closed 2022-06-15 by tgingold