ghdl/ghdl #2096: Shared variable as default value in subprogram

avelure, 2022-06-15:
**Description**
Using a shared variable as default value in subprogram does not provide a warning or error in GHDL, but it does in modelsim and nvc (though not with Aldec). As I understand the default value of an interface object needs to be globally static, which a shared variable is not considered as, though maybe my interpretation is wrong.

**Expected behaviour**
It should generate an error, however it should be a warning with -frelaxed or it will brake most of UVVM.

**How to reproduce?**
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package test_pkg is
  shared variable shared_var : boolean  := false;
  procedure log(
    va1    : boolean    := shared_var
  );
end package;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=93 ent.vhd
```

**Context**
Please, provide the following information:

- OS:
- Origin: Released binaries: nightly e15629320e954921f1332bfe7c97e3fc18f52832

tgingold, 2022-06-15:
I don't think that this is clearly written in the LRM.  It appears simply in the grammar.

JimLewis, 2022-06-15:
@avelure   You are asking about a shared variable of an ordinary type.   You do understand that these were deprecated and removed by VHDL-2002 right?   

What do you expect a simulator to do in the "-frelaxed" and the value is not static.  For example, do you expect it to use the value `false` since that is the value that exists when the package is analyzed or do you expect it to use the value of shared_var when the subprogram actually runs.     Have you checked that this behavior to be the same in different simulators?

avelure, 2022-06-15:
@JimLewis I am well aware that it is deprecated, as for the example being compiled for 93 as you can see.

I was merely questioning discrepancies in error reporting between the simulators. As GHDL normally is of the stricter ones, I was wondering if the missing warning/error was an oversight, a missing check or a misinterpretation of the LRM by the other simulators maybe. If this is in fact undefined from an LRM point of view, I leave it up to the judgement of @tgingold.

I'm not looking for a change in behavior, just in reporting, the simulator is operating as expected. Personally I would expect it to have the value at the time the subprogram runs, as if it was a parameter to the subprogram. This is the behavior modelsim and GHDL currently has. As such, ghdl is already operating as if `-frelaxed` was used.

nickg, 2022-06-15:
Even with protected types you can still end up with very odd behaviour if you allow non-static default expressions. Like the following:

```vhdl
entity static is
end entity;

architecture test of static is

    type counter is protected
        impure function fetch_and_add(n : natural) return natural;
    end protected;

    type counter is protected body
        variable cnt : natural;

        impure function fetch_and_add(n : natural) return natural is
        begin
            cnt := cnt + n;
            return cnt;
        end function;
    end protected body;

    shared variable s : counter;

    function a_pure_function(n : natural := s.fetch_and_add(1)) return integer is
    begin
        return n * 2;
    end function;

    function another_pure_function return integer is
    begin
        return a_pure_function + 1;
    end function;

begin

    p1: process is
    begin
        report integer'image(another_pure_function);
        report integer'image(another_pure_function);
        wait;
    end process;

end architecture;
```

Aldec prints 3, 3 and GHDL prints 3, 5.

tgingold, 2022-06-16:
That's pretty convincing, but this odd behaviour happens because `another_pure_function` is in fact not pure: it calls fetch_and_add which is impure.

JimLewis, 2022-06-16:
@avelure   Just so we are clear, deprecated and removed.   If you are looking for language compliance, a compliant tool would fail with an error if you use a shared variable with an ordinary type if the language revision is 2002 or later.   Have you reported this behavior as a bug to those vendors?

avelure, 2022-06-16:
@jimlewis we both know it is a fruitless effort with this vendor, besides I don't have any avenue of reporting bugs at the moment, but feel free. They will likely tell you that you can upgrade the warning to an error in the tool yourself, if you are so inclined. I prefer GHDL for compliance checking.