ghdl/ghdl #2100: Support subtype as arg `size_res` of to_sfixed, to_ufixed, etc.

umarcor, 2022-06-17:
According to https://github.com/ghdl/ghdl/blob/a952de22c794dc475bb5752378426cf3ec07fdd8/libraries/ieee2008/fixed_generic_pkg-body.vhdl#L3583-L3605

`to_sfixed` can accept an `UNRESOLVED_sfixed` as argument `size_res` (in order to extract the size/range from it). That works if the `UNRESOLVED_sfixed` is a signal or a variable. However, GHDL complains when a subtype is used. See the reproducer below.

```vhdl
library IEEE;
context IEEE.IEEE_std_context;

package my_fixed_pkg is new IEEE.fixed_generic_pkg;

--!

library IEEE;
context IEEE.IEEE_std_context;

library work;
use work.my_fixed_pkg.all;

entity tb_fixed is
end;

architecture arch of tb_fixed is

begin

  process
    subtype stype is sfixed(7 downto -8);
  begin
    report to_string(to_real(stype(to_slv(to_sfixed(-9.96484375, stype)))));
    wait;
  end process;

end;
```

```
ghdl -a --std=08 ent.vhd 
ent.vhd:24:66:error: subtype "stype" not allowed in an expression
    report to_string(to_real(stype(to_slv(to_sfixed(-9.96484375, stype)))));
                                                                 ^
ent.vhd:24:52:error: no overloaded function found matching 'to_sfixed'
    report to_string(to_real(stype(to_slv(to_sfixed(-9.96484375, stype)))));
                                                   ^
ghdl:error: compilation error
```

To my understanding, the subtype and the signal/variable do provide the same information. Hence, I dunno whether this is an "stupid limitation" which might be "fixed" by changin `size_res` to be `constant` rather than the default `variable`.

/cc @JimLewis @Paebbels

umarcor, 2022-06-17:
I think that the same issue might apply to resize in numeric_std:

https://github.com/ghdl/ghdl/blob/a952de22c794dc475bb5752378426cf3ec07fdd8/libraries/ieee2008/numeric_std.vhdl#L1120-L1124

```vhdl
library ieee;
context ieee.ieee_std_context;

entity ent is
  port (
    din  : in signed(15 downto 0);
    dout : out signed(31 downto 0)
  );
end ent;

architecture arch of ent is

begin

  dout <= resize(din, dout'subtype);

end architecture;
```

```vhdl
library ieee;
context ieee.ieee_std_context;

entity ent is
  port (
    din  : in unsigned(15 downto 0);
    dout : out unsigned(31 downto 0)
  );
end ent;

architecture arch of ent is

begin

  dout <= resize(din, dout'subtype);

end architecture;
```

```sh
# ghdl -a --std=08 ent.vhd 
ent.vhd:45:28:error: 'subtype attribute not allowed in an expression
  dout <= resize(din, dout'subtype);
                           ^
ent.vhd:45:17:error: no overloaded function found matching 'resize'
  dout <= resize(din, dout'subtype);
                ^
ghdl:error: compilation error
```

tgingold, 2022-06-18:
The function expect an expression and a subtype is not an expression.

diogratia, 2022-06-18:
It's possible to create an expression that is a value of a specified type qualified expression that's an aggregate:

```vhdl
report to_string(to_real(stype(to_slv(to_sfixed(-9.96484375, stype'(others => '0') )))));
```

closed 2022-08-11 by tgingold