ghdl/ghdl #2101: ghdl does not accept 'high attribute of ranged integer type

akohlsmith, 2022-06-18:
**Description**
Using the high type attribute on a ranged integer type produces an error during analysis:
```
  ghdl -a --std=02 test.vhd
  error: object prefix must be an array
    reset_ctr := reset_ctr'high;
```

**Expected behaviour**
The expected behaviour is to not produce an error; type'high should return the highest value of the type.

**How to reproduce?**
```vhd :file: ent.vhd
entity ent is
    clk: in std_logic;
    reset: in std_logic;
end entity;

architecture a of ent is
begin
  foo: process(clk, reset)
  variable counter: integer range 0 to 15;
  begin
    if reset = '1' then
        counter := counter'high;
    elsif rising_edge(clk) then
        counter := counter - 1;
    end if;
  end process;
end;
```

**Context**
Please, provide the following information:

- OS: OSX 10.15.7
- Origin:
  - homebrew built from source: `GHDL 3.0.0-dev (2.0.0.r151.g176b1453) [Dunoon edition]`

tgingold, 2022-06-19:
GHDL is correct.  For a scalar range, the prefix must be a type or a subtype.  Otherwise, the prefix must be an array.

akohlsmith, 2022-06-19:
hm. if I want to avoid creating a subtype to use `'high`, is there a cleaner way to work with the max value of a ranged integer, or is subtyping it the recommended way?

tgingold, 2022-06-19:
Subtyping is the recommended way, but with vhdl 2008 you can also use 'subtype

closed 2022-06-24 by tgingold