ghdl/ghdl #2103: **GHDL crash** : are_trees_equal: cannot handle IIR_KIND_SUBTYPE_DECLARATION (*std_standard*:1:1)

bellaz89, 2022-06-20:
**Description**

During the execution of a test bench crashes with the following error:

```
are_trees_equal: cannot handle IIR_KIND_SUBTYPE_DECLARATION (*std_standard*:1:1)
```
(See **Context section** for the full program error)

**Expected behaviour**
GHDL either gives an appropriate description of the issues or finishes correctly 

**How to reproduce?**

```bash
  ghdl -a --work=desy --std=93  pkg_math_utils.vhd
  ghdl -a --work=desy --std=93  pkg_logic_misc.vhd     
  ghdl -a --work=desy --std=93  pkg_math_signed.vhd
```
**pkg_math_utils.vhd**
```vhd :file: pkg_math_utils.vhd
------------------------------------------------------------------------------
--          ____  _____________  __                                         --
--         / __ \/ ____/ ___/\ \/ /                 _   _   _               --
--        / / / / __/  \__ \  \  /                 / \ / \ / \              --
--       / /_/ / /___ ___/ /  / /               = ( M | S | K )=            --
--      /_____/_____//____/  /_/                   \_/ \_/ \_/              --
--                                                                          --
------------------------------------------------------------------------------
--! @copyright Copyright 2022 DESY
--! SPDX-License-Identifier: CERN-OHL-W-2.0
------------------------------------------------------------------------------
--! @date 2022-04-01
--! @author Michael Buechler <michael.buechler@desy.de>
--! @author Lukasz Butkowski <lukasz.butkowski@desy.de>
------------------------------------------------------------------------------
--! @brief
--! Math utilities
------------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

--! Package of mathematical utilities and support functions
package math_utils is

  --! Saturation status after an operation.
  type t_saturation is (ST_SAT_OK, ST_SAT_OVERFLOWN, ST_SAT_UNDERFLOWN);

  --! The encoding for _t_staturation_ is
  --!
  --! * ST_SAT_OK         = "00" (No saturation)
  --! * ST_SAT_OVERFLOWN  = "10" (Operation has overflown)
  --! * ST_SAT_UNDERFLOWN = "01" (Operation has underflown)
  attribute enum_encoding : string;
  attribute enum_encoding of t_saturation : type is "00 10 01";

  --! f_bit_length:
  --! Comparable to Python's int.bit_length(),
  --! but when arg is negative, calculate for two's complement.
  --! Attention: returns 1 for argument -1.
  --! When arg is zero, return 0.
  function f_bit_length (arg: integer) return integer;

  --! f_unsigned_length:
  --! Calculates the minimum unsigned signal length to store **arg**,
  --! When arg is zero, return 0.
  function f_unsigned_length (arg: natural) return natural;
[... 581 lines cut ...]
- OS: Ubuntu 20.04.4 LTS
- Origin:
  - [ ] Package manager:
  - [ ] Released binaries:
  - [x] Built from sources: `e15629320e954921f1332bfe7c97e3fc18f52832`
- **Backend** : mcode (default)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/bellandi/Projects/gitlab.msktools.desy.de/fpgafw/lib/desy_vhdl/sim/math/pkg_math_signed/tb/
Command line:
ghdl -a --work=desy --std=93 ../../../../hdl/math/pkg_math_signed.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x55ea98bffe6e 0x55ea98cca325 0x55ea98cc961e 0x55ea98cc9820 0x55ea98cc9466 0x55ea98cc973c 0x55ea98cca356 0x55ea98ccb23b 0x55ea98cc0608 0x55ea98cc08a7 0x55ea98ccd148 0x55ea98cce29f 0x55ea98c9a856 0x55ea98d7d0ce 0x55ea98c8dad5 0x55ea98e43911 0x55ea98a25d1f 0x7f6e57b06081 0x55ea98a242bc 0xfffffffffffffffe
******************************************************************
```

bellaz89, 2022-06-20:
It turns out that the crash doesn't happen if the definition of **f_resize_sat** in package **math_signed** of **pkg_math_signed.vhd** is commented out

bellaz89, 2022-06-20:
It seems that the problem is that GHDL doesn't recognize that the declaration and implementation interface definitions of *function f_resize_sat* (*pkg_math_signed.vhd*) differs. In the former *length* is _positive_, in the latter *length* is _natural_.

tgingold, 2022-06-20:
Yes, it crashes before reporting the error.  I am testing a fix.

commit 12f428f2 (2022-06-21): vhdl-sem.adb: avoid a crash on conformance error. Fix #2103
closed 2022-06-21 by tgingold