ghdl/ghdl #2107: Problems passing ghdl binary for PoSh precompile scripts / missleading documentation

SalttownUser, 2022-06-22:
**Description**

Hi.

I'm using the PowerShell precompile scripts to compile Intel and Lattice vendor libraries.

Prior to d890f27e480cdb37428f1b7c4bb37475ed38348d the environment variable `GHDL` as well as the `-GHDL` parameter had to contain the path to the GHDL
binary **directory** (e.g. `<myGHDLdir>/bin/`). Now, you have to pass the path to the GHDL **executable** (e.g. `<myGHDLdir>/bin/hgdl.exe`). 

The [docstring / help](https://github.com/ghdl/ghdl/blob/a4f663eb442310aa17d895d267a9e212c54c5efe/scripts/vendors/compile-intel.ps1#L84) tells me to pass the binary directory:

~~~
    -GHDL <String>
        Set GHDL binary directory.
~~~

but returns an error when I do:

~~~
.\compile-intel.ps1 -All -GHDL <mypath>\GHDL\MINGW32-mcode-standalone\bin
Use adv. options '-GHDL' to set the GHDL executable.
~~~

The same is true for the bash scripts where help says:

~~~
Advanced options:
  --ghdl <GHDL binary>         Path to GHDL's executable, e.g. /usr/local/bin/ghdl
~~~

But the scripts (e.g. `compile-intel.sh`) give an error message when passing an directory:

~~~
[ERROR] Found GHDL environment variable, but '<mypath>/GHDL/MINGW32-mcode-standalone/bin/' is not a file.
~~~

The related function for the PoSh scripts is [`Get-GHDLBinary`](https://github.com/ghdl/ghdl/blob/778ee5cbb8f7a936f8d4cc93ec6b4671bec2008e/scripts/vendors/shared.psm1#L137)

**Expected behaviour**

When the docs tell me to pass the binary directory, the scripts should work when I do so.

So I would suggest to fix the docstring, or even better to accept both, passing the directory or the executable.

If it is not wanted to make both variants work, I would like to update the error message for the PoSh scripts to something
like `This is not a path to a file. Please pass the path to the GHDL executable e.g. "<mypath>/GHDL/bin/ghdl.exe"`.

**How to reproduce?**
Set environment variable `GHDL` or pass `-GHDL` parameter containing a path to the binary directory to any
precompile script (e.g. `compile-intel.ps1`).

**Context**
Please, provide the following information:

- OS: Win10
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `[tarball_url](https://github.com/ghdl/ghdl/releases/download/nightly/MINGW32-mcode-standalone.zip)`
  - [ ] Built from sources: `commit SHA`

umarcor, 2022-06-23:
@SalttownUser, since you are looking into this topic, this is just to provide some context for you to know:

- There are bash and powershell variants of the precompile scrips.
- However, the implementation in powershell was updated recently compared to bash. Precisely, powershell uses the files specifying the compilation order provided by OSVVM or UVVM, while bash scripts do still have a hardcoded list.
- That is because of the poor features available in bash for handling structs, lists of lists, etc.
- The precompile scripts are used/tested in [ghdl/extended-tests](https://github.com/ghdl/extended-tests).
- In the CI of extended-tests, Action [ghdl/setup-ghdl-ci](https://github.com/ghdl/setup-ghdl-ci) is used. As you can see in https://github.com/ghdl/setup-ghdl-ci/blob/master/main.js#L105-L109, GHDL and GHDL_PREFIX are set by the Action.
- In the mid term, we expect to replace the bash and poweshell scripts with a solution based on Python. See [edaa-org.github.io](https://edaa-org.github.io/). I.e. the way GHDL, GHDL_PREFIX and PREFIX are used might/will have similarities to pyGHDL (https://github.com/ghdl/ghdl/blob/master/pyGHDL/libghdl/__init__.py#L90-L102), thus, the reference to #1557 in #2108.

SalttownUser, 2022-06-24:
Given all this information I think we only should fix the docstring and update the error output of the PoSh script. We should not add functionality that may causes problems somewhere else.

PR #2108 (2022-07-03): Fix docstring and a small bug in PoSh precompile script
**Update**

As discussed [here](https://github.com/ghdl/ghdl/pull/2108#discussion_r906032565) and [here](https://github.com/ghdl/ghdl/issues/2107#issuecomment-1165541878) I changed the PR to only fix the docstring in the preocompile scripts and a small bug in `shared.psm1`.

**Description**

- 290b4d41ba91ee4fb4af91ba8d92108ea8f391ce fixes the try/catch when calling `Get-Command`. Before, the catch condition
   is never met what leads to this error message, when `ghdl..exe` is not on PATH:

   ```
   Get-Command: The term 'ghdl.exe' is no [... cut]
closed 2022-07-03 by Paebbels