ghdl/ghdl #2109: synth: invalid verilog output for null-ranges

bewimm, 2022-06-23:
**Description**
I have some null-range somewhere in my code and that results in illegal verilog output.

This code 
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity bug is
generic(
    tmp : std_ulogic_vector(0 downto 1) := ""
);
port(
    val : out std_ulogic_vector(0 downto 1)
);
end entity;

architecture rtl of bug is
begin
   val <= tmp;
end architecture;
```
produces (`ghdl --synth --out=verilog bug.vhd -e bug`) this illegal verilog:
```
module bug
  (output val);
  localparam n1_o = 0'b;
  assign val = n1_o;
endmodule
```

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `a952de22c`

tgingold, 2022-06-24:
I am not sure what should be done here.  Simply remove nul ports and wire ?

bewimm, 2022-06-24:
I think that makes sense.

I'm not sure if that should also be done for top level ports but at least in my case removing the port would be an acceptable solution.

tgingold, 2022-06-27:
This specific example should be fixed, but I fear additional work will be needed.

commit 928588ec (2022-06-27): netlists-disp_verilog: do not display ports of width 0.  Fix #2109
closed 2022-06-27 by tgingold