ghdl/ghdl #2113: synth: invalid verilog output for null ranges (followup for  #2109)

bewimm, 2022-06-27:
**Description**
As suspected here https://github.com/ghdl/ghdl/issues/2109#issuecomment-1166900650 the previous fix does not cover all such cases.

The following code is closer to my actual use case and it still results in invalid verilog output:
```
library IEEE;
use IEEE.std_logic_1164.all;

entity a is
	port(
		irq : out std_ulogic
	);
end a;

library IEEE;
use IEEE.std_logic_1164.all;

entity b is
	generic(
		NUM_CHANNELS : positive := 4
	);
	port(
		src_channel :  in integer range 0 to NUM_CHANNELS-1;
		src_valid :  in std_ulogic;
		src_ready : out std_ulogic
	);
end b;

architecture struct of a is

	signal src_valid       : std_ulogic;
	signal src_ready       : std_ulogic;
begin
	u0 : entity work.b
	generic map(
		NUM_CHANNELS => 1
	)
	port map(
		src_channel => 0,
		src_valid   => src_valid,
		src_ready   => src_ready
	);
end architecture;

architecture behav of b is
begin
	process(all)
		variable ready         : std_ulogic;
		variable channel_ready : std_ulogic;
	begin
		ready := '1';
		for i in 0 to NUM_CHANNELS-1 loop
			if i = src_channel and src_valid = '1' then
				channel_ready := '0';
			else
				channel_ready := '1';
			end if;
			ready := ready and channel_ready;
		end loop;

		src_ready <= ready;
	end process;

end architecture;
```
The output of ``` ghdl --synth --out=verilog --std=08 a.vhd -e a ``` is
```
module b_1
  (input  src_valid,
   output src_ready);
  wire n11_o;
  wire n14_o;
  wire n16_o;
  assign src_ready = n16_o;
  /* a.vhd:48:44  */
  assign n11_o = 1'b1 & src_valid;
  /* a.vhd:48:25  */
  assign n14_o = n11_o ? 1'b0 : 1'b1;
  /* a.vhd:53:40  */
  assign n16_o = 1'b1 & n14_o;
endmodule

module a
  (output irq);
  wire src_valid;
  wire u0_src_ready;
  localparam n4_o = 1'bZ;
  assign irq = n4_o;
  /* a.vhd:26:16  */
  assign src_valid = 1'bX; // (signal)
  /* a.vhd:29:9  */
  b_1 u0 (
    .src_channel(n1_o),
    .src_valid(src_valid),
    .src_ready());
endmodule

```

**Expected behaviour**
I expect ```src_channel``` to be removed from u0.

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `c060840d41`

tgingold, 2022-06-27:
Now fixed (but may not yet be finished!)

closed 2022-06-27 by tgingold

bewimm, 2022-06-27:
Now the output is 
```
module b_1
  (input  src_valid,
   output src_ready);
  wire n12_o;
  wire n15_o;
  wire n17_o;
  assign src_ready = n17_o;
  /* a.vhd:48:44  */
  assign n12_o = 1'b1 & src_valid;
  /* a.vhd:48:25  */
  assign n15_o = n12_o ? 1'b0 : 1'b1;
  /* a.vhd:53:40  */
  assign n17_o = 1'b1 & n15_o;
endmodule

module a
  (output irq);
  wire src_valid;
  wire u0_src_ready;
  localparam n4_o = 1'bZ;
  localparam n5_o = 0'bX;
  assign irq = n4_o;
  /* a.vhd:26:16  */
  assign src_valid = 1'bX; // (signal)
  /* a.vhd:29:9  */
  b_1 u0 (
    .src_valid(src_valid),
    .src_ready());
endmodule

```
Unfortunately ```localparam n5_o = 0'bX;``` is still not valid verilog.

bewimm, 2022-06-28:
Thanks, it works as expected now.

bewimm, 2022-07-07:
Changing the direction of the null range port to out still tries to map it when the module is instantiated:

```
library IEEE;
use IEEE.std_logic_1164.all;

entity a is
	port(
		irq : out std_ulogic
	);
end a;

library IEEE;
use IEEE.std_logic_1164.all;

entity b is
	generic(
		NUM_CHANNELS : positive := 4
	);
	port(
		src_channel :  out integer range 0 to NUM_CHANNELS-1;
		src_valid :  in std_ulogic;
		src_ready : out std_ulogic
	);
end b;

architecture struct of a is

	signal src_channel     : integer range 0 to 0;
	signal src_valid       : std_ulogic;
	signal src_ready       : std_ulogic;
begin
	u0 : entity work.b
	generic map(
		NUM_CHANNELS => 1
	)
	port map(
		src_channel => src_channel,
		src_valid   => src_valid,
		src_ready   => src_ready
	);
end architecture;

architecture behav of b is
begin
	process(all)
		variable ready         : std_ulogic;
		variable channel_ready : std_ulogic;
	begin
		ready := '1';
		for i in 0 to NUM_CHANNELS-1 loop
			if i = src_channel and src_valid = '1' then
				channel_ready := '0';
			else
				channel_ready := '1';
			end if;
			ready := ready and channel_ready;
		end loop;

		src_ready <= ready;
		src_channel <= NUM_CHANNELS-1;
	end process;

end architecture;
```

```
module b_1
  (input  src_valid,
   output src_ready);
  wire n14_o;
  wire n17_o;
  wire n19_o;
  assign src_ready = n19_o;
  /* bug.vhd:49:44  */
  assign n14_o = 1'b1 & src_valid;
  /* bug.vhd:49:25  */
  assign n17_o = n14_o ? 1'b0 : 1'b1;
  /* bug.vhd:54:40  */
  assign n19_o = 1'b1 & n17_o;
endmodule

module a
  (output irq);
  wire src_valid;
  wire u0_src_channel;
  wire u0_src_ready;
  localparam n5_o = 1'bZ;
  assign irq = n5_o;
  /* bug.vhd:27:16  */
  assign src_valid = 1'bX; // (signal)
  /* bug.vhd:30:9  */
  b_1 u0 (
    .src_valid(src_valid),
    .src_channel(),
    .src_ready());
endmodule
```

```src_channel``` should not show up in the above output either.

tgingold, 2022-07-10:
So it is now fixed.

As far as I understand, all the problem with null ranges have been fixed.

bewimm, 2022-07-10:
Thank you, I haven't run into any other issues with it so far.

tgingold, 2022-10-11:
Yes, you're right.  Now fixed.

commit ffdec03a (2022-06-27): synth/netlists-disp_verilog: skip null input port.  Fix #2113
commit c4f2d830 (2022-07-04): netlists-disp_verilog: handle Id_Abs.  Fix #2113