ghdl/ghdl #2114: Command for displaying code generation backend (--disp-backend)

bjourne, 2022-06-28:
I'm working on some build scripts for ghdl and I'm running into trouble since the backends behave differently. In particular the mcode backend doesn't generate any object files. So it would be nice to be able to call `ghdl --disp-backend` so that you could setup the build differently depending on whether the command prints mcode, gcc, or llvm.

umarcor, 2022-06-28:
FTR, the backend is shown in `ghdl version`. Several tools parse the output to detect the backend. See, for instance, https://github.com/VUnit/vunit/blob/master/vunit/sim_if/ghdl.py#L116-L144.

I'm not arguing against `--disp-backend`, but just providing a workaround.

bjourne, 2022-06-28:
I read through the whole manual but didn't see that it was in the --version command. :) But I think a dedicated  command could be a good idea still.