ghdl/ghdl #2117: GHDL compilation crash where error is expected

Farmadupe, 2022-06-30:
The below code is not valid VHDL because "missing_identifier" is not defined. 

**Expected behaviour**
Expect GHDL to produce an error when analyzing the below code. GHDL actually crashes.

**How to reproduce?**

```vhd :file: bug.vhd
entity bug is end; 

architecture a of bug is 
    type t1 is (enum_val_1);

    procedure p is 
    begin
        enum_val_1.missing_identifier;
    end;
begin 
end;
```

```sh 
ghdl-llvm -a bug.vhd
```

**Context**
Please, provide the following information:

Ubuntu 20.04 GHDL-LLVM, from system repositories.
(Also occurs in a recent ghdl-mcode on Windows (nightly... within the last ~3 months))

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/t/ghdl-bug/bug_2/
Command line:
/usr/lib/ghdl/llvm/ghdl1-llvm -P/usr/lib/ghdl/llvm/vhdl/ieee/v93/ -P/usr/lib/ghdl/llvm/vhdl/ -c -fpic -o bug.o bug.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x4b327a 0x53ed45 0x544497 0x5514d9 0x537501 0x537643 0x57537c 0x567f3c 0x568175 0x538e5f 0x570f65 0x577d39 0x52f5d2 0x6492dc 0x64aedd 0x64c781 0x408fed 0x7fad2f935081 0x40818c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2022-06-30:
the version you are using (0.37) is quite old.  The bug doesn't reproduce on current versions:

ghdl -a bug.vhdl 
bug.vhdl:8:9:error: enumeration literal enum_val_1 cannot be selected by name
        enum_val_1.missing_identifier;
        ^

Farmadupe, 2022-06-30:
Oh I apologize. Although I know the ubuntu repo version of GHDL is ancient, I had actually first encountered this bug at my work environment (windows-mcode nightly from ~February 2022). I assumed that the bug would not have been fixed since February.

If I encounter any further bug, I will make sure I can repro against master GHDL in the future.

closed 2022-06-30 by Farmadupe