ghdl/ghdl #2121: Unknown command `--gen-depends`

LRDPRDX, 2022-07-04:
**Description**
The documentation mention the `--gen-depends` command. Here is the excerpt:

> 
> Generate a Makefile containing only dependencies to build a design unit.
> 
> This command works like the make and gen-makefile commands (see [-m](https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-m)), but instead of a full makefile only dependencies without rules are generated on the standard output. These rules can then be integrated in another Makefile.

But the invocation gives the following error:

```
> ghdl --gen-depends --std=08 and2_tb
ghdl:error: unknown command '--gen-depends', try 'help'

```

Moreover, there is no such a command in the `--help` output:

```
> ghdl --help | grep -e --gen
  alias: --gen-makefile
```

**Expected behaviour**
Generate a Makefile containing only dependencies to build a design unit.

**How to reproduce?**

```
ghdl --gen-depends
```

**Context**

- OS: Linux
- Origin:
  - [ ] Built from sources: downloaded from here: [https://github.com/ghdl/ghdl/archive/master.tar.gz](https://github.com/ghdl/ghdl/archive/master.tar.gz)

tgingold, 2022-07-04:
Yes, --gen-depends is available with gcc/llm backends but not with the mcode backend.

LRDPRDX, 2022-07-04:
Ah, thank you.

commit 470df2c9 (2022-07-04): doc: --gen-depends is not available with mcode.  Fix #2121
closed 2022-07-04 by LRDPRDX