ghdl/ghdl #2122: Synthesis no longer works with specified libraries

efykse, 2022-07-04:
**Description**
Synthesis no longer works while specifying libraries with "--work". Example: ` ghdl --synth -i --work=test_lib test_pkg.vhd test.vhd -e test`

This used to work in `GHDL 2.0.0-dev (v1.0.0-950-g8d512a44)`.

**Expected behaviour**
I expect the synthesized output, instead I get a prompt saying `init>`.

**How to reproduce?**

```vhd :file: test_pkg.vhd
library ieee;
use ieee.std_logic_1164.all;

package test_pkg is
end package test_pkg;

package body test_pkg is
end package body test_pkg;
```
```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;

library test_lib;
use test_lib.test_pkg.all;

entity test is
end test;

architecture rtl of test is
begin
end rtl;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth -i --work=test_lib test_pkg.vhd test.vhd -e test
```

**Context**
- OS: Linux 3.10.0 from `hdlc/formal:all`
- Origin:
  - [x] Built from sources: `GHDL 3.0.0-dev (v2.0.0-402-gbf0f834e8) [Dunoon edition]`

**Additional context**
Console output:
```
# ghdl --synth -i --work=test_lib test_pkg.vhd test.vhd -e test
init>
```

tgingold, 2022-07-04:
Remove `-i`, it's for interactive.

efykse, 2022-07-04:
Thanks for the quick reply! That helped.