ghdl/ghdl #2125: Error in synthesis of latches (constant x)

cderrien, 2022-07-04:
Attempts to synthesise latches may result in constant x in synthesis output.

**Expected behaviour**
I expect the log of a pre-synthesis and post-synthesis simulation to be identical. Failing that, I expect an error or a warning.
Also see details at the end.

**How to reproduce?**
Run the MWE below.

```vhd :file: afed.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity afed is
  port (
    sig : in  std_logic;
    ack : in  std_logic;
    fe  : out std_logic
  );
end entity;

architecture behaviour of afed is
  
  signal fe_h : std_logic;
  signal fe_l : std_logic;
  
begin
  
  fe_h <= '0' when ack = '1' else
          '1' when sig = '1';
  fe_l <= '0' when ack = '1' else
          '1' when sig = '0' and fe_h = '1';
  
  fe <= fe_l;
  
end architecture;
```

```vhd :file: afed_tb.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity afed_tb is
end entity;

architecture behaviour of afed_tb is

  signal sig : std_logic := '1';
  signal ack : std_logic := '1';
  signal fe  : std_logic;
begin

  monitor : process (fe)
  begin
    report std_logic'image(fe);
  end process;

  simu : process
  begin
    ack <= '0'; wait for 1 ns;
    sig <= '1'; wait for 1 ns;
    sig <= '0'; wait for 1 ns;
    sig <= '1'; wait for 1 ns;
    ack <= '1'; wait for 1 ns;
    wait;
  end process;

  afed : entity work.afed
    port map (
      sig => sig,
      ack => ack,
      fe  => fe
      );
  
end architecture;
```

```sh :file: run.sh
ghdl -a afed.vhdl
ghdl --synth --latches --out=vhdl afed > afed.synth.vhdl
ghdl -a afed_tb.vhdl
ghdl -e afed_tb
./afed_tb > log_sim.txt
rm -f *.cf
ghdl -a afed.synth.vhdl
ghdl -a afed_tb.vhdl
ghdl -e afed_tb
./afed_tb > log_syn.txt
diff -y log_sim.txt log_syn.txt | tee log_diff.txt
```

**Context**
Running on ubuntu 20, ghdl build from source.
The hash of the last commit is 0529da6ce3ebe08e2f01f5916400adb2b9344bf8

**Additional context**
I noticed that more straightforward latches are synthesised correctly. For example, splitting the fe_h and fe_l into their own entities produces non-x output (though I did not check to see if it was correct).
Looking at src/synth/netlist-inference.adb, it looks like some false loops are wrongly detected.
So Walk_Nets does not return false : ghdl thinks this is not a real latch and should be optimised out (this might also explain why splitting the entity does not result in the error occurring).

tgingold, 2022-07-05:
You're right.  I fear there is no immediate fix, I have to investigate more.

Xiretza, 2022-07-05:
Ref #1501

cderrien, 2022-07-05:
I modified the MWE after looking at the issue Xiretza linked, and it turns out it works with some modification.

```vhd :file: afed.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity afed is
  port (
    sig : in  std_logic;
    ack : in  std_logic;
    fe  : out std_logic
  );
end entity;

architecture behaviour of afed is
  
  signal fe_h : std_logic;
  --signal fe_l : std_logic;
  
begin
  
  fe_h <= '0' when ack = '1' else
          '1' when sig = '1';
  fe   <= '0' when ack = '1' else
          '1' when sig = '0' and fe_h = '1';
  
  --fe <= fe_l;
  
end architecture;
```
```vhd :file: afed_tb.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity afed_tb is
end entity;

architecture behaviour of afed_tb is

  signal sig : std_logic := '1';
  signal ack : std_logic := '1';
  signal fe  : std_logic;
begin

  monitor : process (fe)
  begin
    report std_logic'image(fe);
  end process;

  simu : process
  begin
    wait for 1 ns;
    ack <= '0'; wait for 1 ns;
    sig <= '1'; wait for 1 ns;
    sig <= '0'; wait for 1 ns;
    sig <= '1'; wait for 1 ns;
    ack <= '1'; wait for 1 ns;
    wait;
  end process;

  afed : entity work.afed
    port map (
      sig => sig,
      ack => ack,
      fe  => fe
      );
  
end architecture;
```
Same commands as original post above.

commit 493d10f3 (2022-07-11): netlists-inference: detect false loops only for variables.  Fix #2125
closed 2022-07-11 by tgingold