ghdl/ghdl #2130: Synthesis errors out even though elaboration works fine

jarvis-lorry, 2022-07-06:
I'm trying to run synthesis ghdl and I get the following error: entity "lfsr" cannot be at the top of a design.  I've included the two vhdl files, which are from opencores.org

I can successfully analyze and elaborate the design with the following two statements:
ghdl -e --std=08 galois-lfsr.vhdl pkg-types.vhdl 
gdhl -e --std=08 lfsr

When I attempt synthesis, it errors out.
ghdl --synth --std=08 --out=verilog --no-formal galois-lfsr.vhdl pkg-types.vhdl -e lfsr

ghdl:error: entity "lfsr" cannot be at the top of a design
galois-lfsr.vhdl:53:17:error: (generic "taps" has no default value)
                taps:boolean_vector

I'm running GHDL 3.0.0

GHDL 3.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 mcode code generator
Written by Tristan Gingold.

[galois-lfsr.vhdl.txt](https://github.com/ghdl/ghdl/files/9058117/galois-lfsr.vhdl.txt)
[pkg-types.vhdl.txt](https://github.com/ghdl/ghdl/files/9058118/pkg-types.vhdl.txt)

umarcor, 2022-07-06:
The same error should be shown if you tried to run/simulate the design. You need to either modify the sources to provide a default value for `taps` or provide it as a cli option `-gtaps=VALUE`. See http://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-gGENERIC.

jarvis-lorry, 2022-07-07:
Thank you, I should have noticed that.

closed 2022-07-07 by umarcor