ghdl/ghdl #2131: case statement does not respect choice values of strings of different length (incorrect length for the choice value)

tristanitschner, 2022-07-10:
Example:
```vhdl
entity incorrect_choice_length is
end entity;

architecture arch of incorrect_choice_length is

type some_datastructure_t is record
	name  : string;
	value : natural range 0 to 3;
end record;

type some_datastructure_array is array (natural range <>) of some_datastructure_t;

constant some_datastructure : some_datastructure_array(0 to 3) := (
	0 => (name => "zarro", value => 0),
	1 => (name => "one",   value => 1),
	2 => (name => "two",   value => 2),
	3 => (name => "three", value => 3)
);

begin

	process is
		variable int : integer;
		procedure printValue(constant x: some_datastructure_t) is
		begin
			case x.name is
				when "zarro" =>
					report "We got value : " & natural'image(x.value);
				when "one" =>
					report "We got value : " & natural'image(x.value);
				when "two" =>
					report "We got value : " & natural'image(x.value);
				when "three" =>
					report "We got value : " & natural'image(x.value);
			end case;
		end procedure;
	begin
		printValue(some_datastructure(0));
		printValue(some_datastructure(1));
		printValue(some_datastructure(2));
		printValue(some_datastructure(3));
	end process;

end architecture;
```

Command and error message:
```sh
$ ghdl analyze --std=08 incorrect_choice_length.vhd
incorrect_choice_length.vhd:29:38:error: incorrect length for the choice value
                                when "one" =>
                                     ^
incorrect_choice_length.vhd:31:38:error: incorrect length for the choice value
                                when "two" =>
                                     ^
```
GHDL version:
```sh
$ ghdl --version
GHDL 3.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 10.3.0
 mcode code generator
Written by Tristan Gingold.
```

tgingold, 2022-07-10:
GHDL is correct, the length of strings in a case statement must be the same.
(And likewise for the record, although this is not yet spotted).

diogratia, 2022-07-11:
If you comment out the process with the case statement and analyze:
```none
%: ghdl -a --std=08 incorrect_choice_length.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r444.ga33efb91d) [Dunoon edition]
Compiled with GNAT Version: 12.1.0
Target: aarch64-apple-darwin21
/Users/dio_gratia/Desktop/
Command line:
/Users/dio_gratia/opt/share/ghdl/bin/ghdl1-llvm --std=08 -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ieee/v08/ -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ -c -o incorrect_choice_length.o incorrect_choice_length.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:79
Load address: 0x1029f0000
Call stack traceback locations:
0x102c8f37c 0x102c91234 0x102ca1190 0x102c9afac 0x102c9c818 0x102c9ce88 0x102c9db88 0x102c9af98 0x102c9b698 0x102c9bb90 0x102c9c4b4 0x102c9b810 0x102c9daa8 0x102c9db74 0x102ccbaf8 0x102ccbe64 0x102ccbf80 0x102cd37c8 0x102cfdda8 0x102cb7f30 0x102cabb08 0x102d0253c 0x102d03f90 0x102d055fc 0x1029f6fe4
******************************************************************
ghdl:error: compilation error
%:
```
we see there's an uncaught internal error.

By amazing (cough!) coincidence if we just declare an unbounded array type:

```vhdl
entity array_of_string_elements is
end entity;

architecture foo of array_of_string_elements is
    type strings is array (natural range <>) of string;
    constant strings_constants: strings(0 to 3) := 
                          ("one", "two", "three", "four");
    
begin
end architecture;
```
We get the very same uncaught internal exception:

```none
%: ghdl -a --std=08 array_of_string_elements.vhdl
array_of_string_elements.vhdl:7:42:warning: expression subtype doesn't match aggregate element subtype [-Wruntime-error]
                          ("one", "two", "three", "four");
                                         ^
array_of_string_elements.vhdl:7:51:warning: expression subtype doesn't match aggregate element subtype [-Wruntime-error]
                          ("one", "two", "three", "four");
                                                  ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r444.ga33efb91d) [Dunoon edition]
Compiled with GNAT Version: 12.1.0
Target: aarch64-apple-darwin21
/Users/dio_gratia/Desktop/
Command line:
/Users/dio_gratia/opt/share/ghdl/bin/ghdl1-llvm --std=08 -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ieee/v08/ -P/Users/dio_gratia/opt/share/ghdl/lib/ghdl/ -c -o array_of_string_elements.o array_of_string_elements.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:79
Load address: 0x10205c000
Call stack traceback locations:
0x1022fb37c 0x1022fd95c 0x1022fbce8 0x1022fc020 0x1022fd990 0x102335c14 0x10233cab0 0x10233e6d8 0x1023624bc 0x10232379c 0x102317b08 0x10236e53c 0x10236ff90 0x1023715fc 0x102062fe4
******************************************************************
ghdl:error: compilation error
%:
```
With a bit more information.

It tells us among other things ghdl doesn't completely support -2008 unbounded array subelement definitions and umarcor's feature request flag is warranted.

You can also find the original error regarding mismatched choice expression array lengths in IEEE-Std 1076-2008 10.9 Case statements:

>If the expression is of a one-dimensional character array type and is not described by either of the preceding two paragraphs, then the values of all of the choices, except the **others** choice, if present, shall be of the same length. Moreover, for an ordinary case statement, or for a matching case statement in which the expression is of an array type whose element type is BIT, each value of the (base) type of the expression shall be represented once and only once in the set of choices, and no other value is allowed. For a matching case statement in which the expression is of an array type whose element type is STD_ULOGIC, each value of the (base) type of the expression, other than an array value containing '–' as an element, shall be represented (as defined in the preceding paragraph) once and only once in the set of choices of the case statement. It is an error if a choice does not represent a value of the (base) type of the expression other than an array value containing '–' as an element. In all cases, it is an error if the value of the expression is not of the same length as the values of the choices. If there is only one choice and that choice is **others**, then the value of the expression may be of any length.

commit 3f44eee9 (2022-08-11): vhdl-sem_expr: fix a crash on invalid aggregate.  Fix #2131