ghdl/ghdl #2133: cocotb: forcing 16 bit inout std_logic_vector port of intel maxii CPLD sim netlist crashes ghdl

besthwengineer, 2022-07-15:
**Description**
Created a sim netlist out of Quartus Prime V21.1 (for Altera Quartus II V13.1 it is the same) using
 - MAX II CPLD
 - Block Diagram / Schematic File (.bdf)
 - 16 bit INOUT address data bus "MC_AD"
 - output enable IN signal "OE_AD" to steer MC_AD direction
 - 16 bit OUT signal "ad_out" to give out if MC_AD is switched to IN
 - internal constant 0xBEEF is forced out if MC_AD is switched to OUT

Setup cocotb with ghdl vendor lib compile script (powershell): compile-intel.ps1

I put the sim netlist directly out of Quartus into cocotb
 - ultra simple testbenching with
```py :file: cocotb_min_bidi_sample_maxii.py
    import cocotb
    from cocotb.triggers import Timer

    @cocotb.test()
    async def min_bidi_sample_maxii_test(dut):
    
        # Init Stimuli-Signale
        dut.OE_AD.value = 0
        dut.MC_AD.value  = 0xFAFA
    
        # um Werte gültig werden zu lassen kurz warten
        await Timer(1, units = "ns")
```

If you start the simple cocotb testbench with make, ghdl just crashes instantly, see bug output below.

**Expected behaviour**
cocotb simulation should just simulate with nothing to happen (PASS) because my cocotb testbench does nothing after initializing IN and INOUT ports

**How to reproduce?**
just use my cocotb example i provide (zip-file). 
The only thing you have to add to my provided cocotb testbench files
 - min_bidi_sample_maxii.vho
 - cocotb_min_bidi_sample_maxii.py
 - Makefile
is to compile the "intel" libraries in the root folder, e.g. with the ghdl vendor powershell compile script "compile-intel.ps1". Out of the intel libraries you only need "maxii".

```vhd :file: min_bidi_sample_maxii.vho
-- This is only an excerpt of the sim VHDL netlist file, the whole file is below
LIBRARY IEEE;
LIBRARY MAXII;
USE IEEE.STD_LOGIC_1164.ALL;
USE MAXII.MAXII_COMPONENTS.ALL;

ENTITY 	min_bidi_sample_maxii IS
    PORT (
	ad_out : OUT std_logic_vector(15 DOWNTO 0);
	OE_AD : IN std_logic;
	MC_AD : INOUT std_logic_vector(15 DOWNTO 0)
	);
END min_bidi_sample_maxii;

-- Design Ports Information

ARCHITECTURE structure OF min_bidi_sample_maxii IS
```
**Context**

- OS: Windows 10
- Origin:
  - [ ] Released binaries: `[tarball_url](https://github.com/ghdl/ghdl/releases/download/nightly/MINGW64-mcode-standalone.zip)`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r483.g1668538d0) [Dunoon edition]
Compiled with GNAT Version: 12.1.0
Target: x86_64-w64-mingw32
C:\projects\python\cocotb_min_bidi_sample_maxii_quartus21\
Command line:
C:\Tools\ghdl_mingw64\bin\ghdl.exe -r -Pintel --ieee=synopsys --workdir=sim_build -Psim_build --work=work min_bidi_sample_maxii --vpi=C:/Tools/miniconda3/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-signals.adb:3522 access check failed
Load address: 0x7ff6a7670000
Call stack traceback locations:
0x7ff6a76da3ce 0x7ff6a76db195 0x7ff6a76c8540 0x7ff6a76c8d49 0x7ff6a76c8def 0x7ff6a767380a 0x7ff6a77d5534 0x7ff6a77d560a 0x7ff6a7a79e4c 0x7ff6a79b231e 0x7ff6a78ca9c7 0x7ff6a7a7a8bb 0x7ff6a767363c 0x7ff6a76713ac 0x7ff6a76714e4 0x7fffccf57032 0x7fffcd24264f
******************************************************************
```
[cocotb_min_bidi_sample_maxii_quartus21.zip](https://github.com/ghdl/ghdl/files/9121101/cocotb_min_bidi_sample_maxii_quartus21.zip)

tgingold, 2022-07-18:
Yes, there are a few issues related to cocotb, I will address them at some point.

dalex78, 2026-09-04:
# Issue #2133 — cocotb writing an `inout std_logic_vector` on a MAX II netlist

Analysis by an AI agent. The scenario works on master, and the code that produced the reported signature was fixed after the report. The crash itself could not be reproduced here, and this says exactly why.

## What was reported

cocotb sets two ports of a Quartus/MAX II simulation netlist —

```python
dut.OE_AD.value = 0
dut.MC_AD.value = 0xFAFA
await Timer(1, units = "ns")
```

— and GHDL 3.0.0-dev (`2.0.0.r483.g1668538d0`) dies immediately:

```
raised CONSTRAINT_ERROR : grt-signals.adb:3522 access check failed
```

*"regardless of the direction of the inout port"*. tgingold's only reply was that there were several cocotb-related issues to address at some point.

## What that line was

`grt-signals.adb:3522` in that era is inside `Update_A_Signal`, in the `Net_One_Driver` branch:

```ada
            --  Update driver
            Trans := Sig.S.Drivers (0).First_Trans.Next;
            if Trans /= null then
               Free (Sig.S.Drivers (0).First_Trans);
               Sig.S.Drivers (0).First_Trans := Trans;
            end if;

            --  Update driving value (unless forced)
            if not Sig.Flags.Is_Drv_Forced then
               case Trans.Kind is
```

`Trans` is tested for null, then dereferenced unconditionally a few lines later. "access check failed" is that dereference: the signal was updated with no pending transaction on its driver, which is what a VPI deposit can produce.

**Master has fixed it.** Commit `2f4884857` ("simul: handle force/release signal assignments", 2023-01-03) moved the whole `if not Sig.Flags.Is_Drv_Forced` block *inside* `if Trans /= null then`, so there is no unguarded dereference any more. The report is from 2022-07-18, so the fix landed about six months later and was not aimed at this issue.

## What was verified here

**cocotb, for real.** cocotb 2.1.0 in a virtualenv, driving a design shaped like the reporter's netlist — a flat entity with `ad_out : out std_logic_vector(15 downto 0)`, `OE_AD : in std_logic` and `MC_AD : inout std_logic_vector(15 downto 0)`, with per-bit tristate buffers instantiated in a generate, as a vendor netlist emits them — and the reporter's own test body. It passes:

```
1.00ns INFO  cocotb.min_bidi   OE_AD=0  MC_AD=1111101011111010  ad_out=1111101011111010
2.00ns INFO  cocotb.min_bidi   OE_AD=1  MC_AD=1011111011101111  ad_out=1011111011101111
** TESTS=2 PASS=2 FAIL=0 SKIP=0 **
```

Both bus directions, since the report says the direction does not matter: with `OE_AD = 0` the deposited `0xFAFA` stands, and with `OE_AD = 1` the design's own `0xBEEF` drivers win.

**The same thing without cocotb**, as `testsuite/vpi/issue2133`: a VPI module that takes handles on the ports, deposits `0xFAFA` into the `inout` vector with `vpi_put_value (…, vpiNoDelay)` and reads it back a time unit later, so a wrong deposit fails on the value and not merely on the absence of a crash. It passes on mcode, llvm and llvm-jit.

## What could not be done

**The MAX II library could not be retrieved.** It is not a downloadable artifact: it ships inside Intel Quartus Prime, and GHDL's own `scripts/vendors/compile-intel.sh` compiles it *from a local Quartus installation* (`$QUARTUS_ROOTDIR`). There is no Quartus here and no way to obtain one without an Intel account and a licence acceptance, so the reporter's `min_bidi_sample_maxii.vho` -- which instantiates MAX II primitives -- cannot be elaborated.

**The crash was not reproduced.** Beyond the netlist-shaped design above, three other shapes were tried against both master and a build of the **`v2.0.0` tag**, which still has the unguarded dereference:

- an `inout` vector never driven inside the entity, only read;
- an `inout` vector with a single whole-vector tristate driver;
- the per-bit tristate version, with the deposit at `OE_AD = 0` and at `OE_AD = 1`.

All eight combinations pass. So the vulnerable code is reachable in principle but not by any of these designs, and what remains untested is what the MAX II primitives do that these do not.

## Recommendation

Ask the reporter to retry on a current GHDL. The dereference that produced their exact signature is guarded now, the scenario their testbench exercises works with real cocotb, and there is a regression test for it. If it still fails, the useful next step is the `GHDL Bug occurred` block from a current build -- the line it names will no longer be `grt-signals.adb:3522` and will point somewhere new.

Do not close it on this analysis alone: nothing here was run against the actual MAX II netlist.

## What this change adds

Nothing in `src/`. `testsuite/vpi/issue2133` is the VPI test described above. The cocotb setup is not added to the testsuite: cocotb is not among GHDL's test dependencies, and the VPI test covers the same mechanism with the harness the suite already has.