ghdl/ghdl #2137: Build error:  gnatmake: ".../ghdl/src/synth/netlists-disp_dot.adb" compilation error

titan73, 2022-07-21:
kmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /prj/usb_4531/users/costa/eda/ghdl/src/synth/netlists-disp_dot.adb
netlists-iterators.ads:25:06: aspect identifier expected
netlists-iterators.ads:28:46: aspect identifier expected
netlists-iterators.ads:46:06: aspect identifier expected
netlists-iterators.ads:49:49: aspect identifier expected
netlists-iterators.ads:64:06: aspect identifier expected
netlists-iterators.ads:67:50: aspect identifier expected
netlists-iterators.ads:83:06: aspect identifier expected
netlists-iterators.ads:86:48: aspect identifier expected
netlists-iterators.ads:100:06: aspect identifier expected
netlists-iterators.ads:103:45: aspect identifier expected
netlists-iterators.ads:117:06: aspect identifier expected
netlists-iterators.ads:120:46: aspect identifier expected
netlists-iterators.ads:134:06: aspect identifier expected
netlists-iterators.ads:136:53: aspect identifier expected
netlists-iterators.ads:149:06: aspect identifier expected
netlists-iterators.ads:152:43: aspect identifier expected
netlists-iterators.ads:166:06: aspect identifier expected
netlists-iterators.ads:169:44: aspect identifier expected
gnatmake: "/prj/usb_4531/users/costa/eda/ghdl/src/synth/netlists-disp_dot.adb" compilation error
make: *** [ghdl_mcode] Error 4

titan73, 2022-07-21:
OS is RH7 64 bit

tgingold, 2022-07-21:
Try with --disable-synth
Your gcc/gnat compiler is too old.

titan73, 2022-07-22:
With --disable-synth, it just further but fails in the same way.
Indeed, my gnat version is older than 8. I thought configure would check gnat version but there's no such check. That would have been nice. Thanks for the answer.

closed 2022-07-22 by titan73

mikaelsky, 2023-03-15:
If it is still relevant @titan73 
It is feasible to get A build of GHDL under RHEL7. The challenge is that even with --disable-synth the flow still builds the netlist builder option to support vhdl-debug. These are the files that are causing issues with older GCC version.

The quick hacky workaround involve:
moving src/synth/netlists-iterators.ad*  and src/synth/netlists-dump.ad* out of src/synth. These are the files that cause issues, specifically the iterators file.

Then vhdl-debug must be disabled. This is achieved by commenting out
-- with Synth.Environment.Debug;
--   package Debug is new Env.Debug;
In src/synth/synth-vhdl_environments.ads

With these changes the latest GHDL builds on rhel7 and passes my simple hello_world example.
$ ghdl --version
GHDL 4.0.0-dev (3.0.0.r17.gea460cb.dirty) [Dunoon edition]
 Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-28
 mcode code generator