ghdl/ghdl #2139: STORAGE_ERROR from plugin for yosys

JulianKemmerer, 2022-07-23:
**Description**
While attempting load a design into yosys via the GHDL plugin I get a storage error.

**Expected behaviour**
I expect the design to successfully parse, continue through yosys synthesis, etc

**How to reproduce?**
Unzip the attached archive, change into directory, run script
https://drive.google.com/file/d/1yklCXDWC6KXMwjUmbAoD5z43tN2tkc6i/view?usp=sharing
```
cd ghdl-storage-error
bash top/top_8239.sh
```
That script invokes a command like `yosys -m ghdl  -p 'ghdl --std=08 file.vhd` , etc 
And produces a log in the `ghdl-storage-error` directory called `open_tools_8239.log`.

That log file ends with:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r506.g8be816aae) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/julian/Desktop/ghdl-storage-error/
Command line:

Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
******************************************************************
ERROR: vhdl import failed.
```

**Context**
- OS: Linux Xubuntu 20.04
- Origin:
I tried both building GHDL+the plugin today.
And tried the 7-23-22 last night build of the OSS CAD Suite from yosyshq.

**Additional context**
Sorry that its a bunch of generated VHDL code and other junk - issue might stem from bad code gen but I can't tell  :weary: 

Thanks for your time!

tristanitschner, 2022-07-24:
I've looked at this a little bit.
First I tried to elaborate the design:
```sh
find . -name "*.vhd" -exec ghdl import --std=08 {} \;
ghdl make --std=08 top_8239
ghdl elaborate --std=08 top_8239
```
This results in a lot of warnings of the form:
```
./bit_math.h/int23_abs/int23_abs_1CLK_8c44261b.vhd:50:12:warning: declaration of "return_output" hides port "return_output" [-Whide]
```
When you then run synthesis, you can see that GHDL creates signals twice, for each of these cases:
```
ghdl synth --std=08 top_8239
```
Of course, the same signal declaration twice is not valid VHDL code.

Also it says in the VHDL 2008 LRM in the section "Scope and Visibilty", page 189 first paragraph, that name overloading is allowed. GHDL synthesis just can't deal with it as of right now.

Furthermore somehow the synthesized output contains the conversion function "unresolved_unsigned", which is only defined for VHDL 2008 and not VHDL 1993, which is the default output format. I don't know what's happening here.

JulianKemmerer, 2022-07-25:
Hmm interesting - I will see if the 'decl hides port' warnings can be avoided via better code gen - I had it in mind that those warnings werent issues - but I should re-evaluate... If I am really ~'declaring a signal twice'/overloading unintentionally that sounds wonky indeed...

Not sure regarding VHDL 93 or 08 in terms of causing issue? I do intended to target 2008 for synthesis assuming that's possible, idk whats going on with unresolved_unsigned in there...

tgingold, 2022-07-25:
Well, a warning is just a warning and shouldn't be a problem.
There was a memory issue, which I think is now fixed.

tgingold, 2022-07-25:
`unsigned_resolved` is the conversion type from vhdl-08.  Maybe this should be adjusted.

JulianKemmerer, 2022-07-25:
Wow thanks for the quick fix! Gosh now I'm less motivated to clean my code gen to fix those warnings :zany_face: 

`type from vhdl-08.  Maybe this should be adjusted.` - is there anything I am doing wrong/could do better in code? I dont understand - is GHDL converting VHDL 08 -> like 93 internally or something like that? ~dont understand 'from 08' if targeting `--std=08` and unsigned needing anything different 08 vs ? etc

Thanks again!

tgingold, 2022-07-25:
Reopen to fix `unsigned_resovled`

tgingold, 2022-07-27:
Should now be fixed.
You did nothing wrong, I adjust the output so that it is vhdl93 compatible.

commit 6f8aaa49 (2022-07-25): synth-environment: fix memory crash.  Fix #2139
commit 1f3162b2 (2022-07-27): synth-disp_vhdl: improve output for unsigned.  Fix #2139
closed 2022-07-27 by tgingold