ghdl/ghdl #2142: "x" is obsoleted by package "y" using last docker image

andresdemski, 2022-07-26:
**Description**
I'm trying to simulate grlib-leon3 processor in a custom environment. I was using  GHDL 1.0.0 and I was able to generate the testbench file. Now I'm trying to simulate that with some other verilog files and I'm trying to use --synth command. In order to do that, I moved everything to use the last ghdl/ghdl:ubuntu-llvm docker image and the building process started to fail whit the following errors:

```
gaisler//leon3v3/libiu.vhd:33:13:error: package "libfpu" is obsoleted by package "gencomp"
gaisler//leon3v3/libleon3.vhd:37:13:error: package "libfpu" is obsoleted by package "leon3"
...
...
```

Before starting the docker, I removed all output paths and clean the working directory. I think nothing is cached from previous run

**Expected behaviour**
Generate verilog output

**How to reproduce?**

```
rm -rf  gnu *.o
mkdir -p gnu/techmap gnu/work gnu/gasiler gnu/grlib
ghdl -i --mb-comments --workdir=gnu/techmap --work=techmap -Pgnu $TECHMAP_SRCS
ghdl -i --mb-comments --workdir=gnu/grlib --work=grlib -Pgnu $GRLIB_SRCS
ghdl -i --mb-comments --workdir=gnu/gaisler --work=gaisler -Pgnu $GRLIB_SRCS
ghdl -i --mb-comments --workdir=gnu/work --work=work -Pgnu $WORK_SRCS
ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments --warn-no-binding -O2 --workdir=gnu/work --work=work -Pgnu -Pgnu/techmap -Pgnu/gaisler -Pgnu/grlib leon3x_wrapper
```

**Context**

- OS: 5.18.10-1-MANJARO
- Origin:
  - [ ] Package manager: Pacman v6.0.1
  - [] Docker image: ghdl/ghdl:ubuntu18-llvm-5.0

Let me know if you need more information.
Regards
Andres

tgingold, 2022-07-26:
`ghdl -i` just imports but do not analyze, while `ghdl --synth` expects a ready to be elaborated unit if you don't pass files.

Either you need to run `ghdl -m` before `ghdl --synth`, or skip `ghdl -i` and pass all the files to `ghdl --synth`.  You can use `--work=` between files, like this: `ghdl --synth ... --work=techmap $TECHMAP_SRCS --work=grlib $GRLIB_SRCS ...`

andresdemski, 2022-07-26:
Thanks! That works but exposes another issue:

```
<WORK_PATH>/leon3x_wrapper.vhd:205:12:warning: signal "ahbso" is never assigned [-Wnowrite]
    signal ahbso : ahb_slv_out_vector := (others => ahbs_none);
           ^
<GRLIB_GAISLER_PATH>/leon3v3/leon3x.vhd:179:11:warning: unhandled attribute "sync_set_reset"
attribute sync_set_reset of rst : signal is "true";
          ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:230:12:warning: no assignment for offsets 0:107 of variable "vitdatain"
  variable vitdatain : itdatain_type;
           ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:231:12:warning: no assignment for offsets 0:107 of variable "vdtdatain"
  variable vdtdatain : dtdatain_type;
           ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:232:12:warning: no assignment for offsets 0:107 of variable "vdtdatain2"
  variable vdtdatain2 : dtdatain_type;
           ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:169:10:warning: no assignment for offsets 0:107 of signal "itdataout"
  signal itdataout : itdataout_type;
         ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:173:10:warning: no assignment for offsets 0:95 of signal "iddataout"
  signal iddataout : iddataout_type;
         ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:193:10:warning: no assignment for offsets 0:107 of signal "dtdataout"
  signal dtdataout : dtdataout_type;
         ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:194:10:warning: no assignment for offsets 0:107 of signal "dtdataout2"
  signal dtdataout2: dtdataout_type;
         ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:199:10:warning: no assignment for offsets 0:95 of signal "dddataout"
  signal dddataout : dddataout_type;
         ^
<GRLIB_GAISLER_PATH>/leon3v3/cachemem.vhd:205:10:warning: no assignment for offsets 0:11 of signal "ddenablev"
  signal ddenablev : cbwmasktype;
         ^
<GRLIB_GAISLER_PATH>/leon3v3/iu3.vhd:110:13:warning: unhandled attribute "sync_set_reset"
  attribute sync_set_reset of rstn : signal is "true"; 
            ^
<GRLIB_GAISLER_PATH>/leon3v3/iu3.vhd:2936:31:warning: comparing non-numeric vector is unexpected
    if (irl = "1111") or (irl > r.w.s.pil) then
                              ^
<GRLIB_GAISLER_PATH>/leon3v3/iu3.vhd:105:5:warning: no assignment for port "tbi_2p" [-Wnowrite]
    tbi_2p : out tracebuf_2p_in_type;
    ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (v2.0.0-502-g341a8698c) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/ademski/novo/novohdl/build/
Command line:
ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments --warn-no-binding -O2 --workdir=gnu/work --work=work -Pgnu -Pgnu/techmap -Pgnu/gaisler -Pgnu/grlib leon3x_wrapper
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:403
Call stack traceback locations:
0x7fa239fc07b1 0x55e324a6d2d0 0x55e324a8f849 0x55e324a90e71 0x55e324bd7f54 0x55e324c489d5 0x55e324c48a22 0x55e324bd7e80 0x55e324ca5d22 0x55e324b87673 0x55e324cae91e 0x55e324a12829 0x7fa239794c85 0x55e324a115e8 0xfffffffffffffffe
******************************************************************
make: *** [Makefile:30: build] Error 2
```

Should i open a new issue for that?

tgingold, 2022-07-27:
Not yet, I think I can reproduce it.

commit ed440ad3 (2022-07-27): netlists-expands: do not try to clean input of dyn_extract.  Fix #2142
closed 2022-07-27 by tgingold