ghdl/ghdl #2144: Invalid bounds check for multidimensional array

bewimm, 2022-07-27:
**Description**
The bounds check for multidimensional arrays seems to use the constraint for the first index for all the other ones.

Reproducer:
```
library IEEE;
use IEEE.std_logic_1164.all;

entity bug is
	port (
		dummy : in std_ulogic
	);
end bug;

architecture struct of bug is
	type table_t is array (natural range<>, natural range<>) of std_ulogic;

	function fun return table_t is
		variable ret : table_t(0 to 1, 0 to 3);
	begin
		return ret;
	end function;

	constant table : table_t := fun;
	
begin
	gen_i : for i in table'range(1) generate
		gen_j : for j in table'range(2) generate
			b : block is
				function print return std_ulogic is
				begin
					report "index="& integer'image(i) & "," & integer'image(j) & "; " &
					       "length="& integer'image(table'length(1)) & "," & integer'image(table'length(2));
					return '0';
				end function;

				constant tmp : std_ulogic := print;
				constant entry : std_ulogic := table(i, j);
			begin

			end block;
		end generate;
	end generate;
end architecture;
```
Command: ```ghdl --synth --out=vhdl --std=08 --work=util bug.vhd -e bug```
Output:
```
bug.vhd:27:41:(report note): index=0,0; length=2,4
bug.vhd:27:41:(report note): index=0,1; length=2,4
bug.vhd:27:41:(report note): index=0,2; length=2,4
bug.vhd:33:69:error: index not within bounds
                                constant entry : std_ulogic := table(i, j);
                                                                    ^
bug.vhd:27:41:(report note): index=0,3; length=2,4
bug.vhd:33:69:error: index not within bounds
                                constant entry : std_ulogic := table(i, j);
                                                                    ^
bug.vhd:27:41:(report note): index=1,0; length=2,4
bug.vhd:27:41:(report note): index=1,1; length=2,4
bug.vhd:27:41:(report note): index=1,2; length=2,4
bug.vhd:33:69:error: index not within bounds
                                constant entry : std_ulogic := table(i, j);
                                                                    ^
bug.vhd:27:41:(report note): index=1,3; length=2,4
bug.vhd:33:69:error: index not within bounds
                                constant entry : std_ulogic := table(i, j);
                                                                    ^
```

tgingold, 2022-07-27:
Oops, you were right.  Now fixed.
Thank you for the issue.

closed 2022-07-27 by tgingold

bewimm, 2022-07-27:
Works now. Thanks for the quick fix as usual!

commit af37a8e1 (2022-07-27): elab-vhdl_expr: fix handling of multi-dim arrays.  Fix #2144