ghdl/ghdl #2151: GHDL mcode runtime crash

Farmadupe, 2022-07-28:
**Description**
The below code causes GHDL-mcode to silently crash at runtime on windows.

**Expected behaviour**
Expect no crash. 
Note: This issue was originally discovered in a different codebase with GHDL-mcode on linux (redhat 8, compiled from source, less than one month old). This causes a compilation crash on **trans-chap3.adb:3318**.

**How to reproduce?**
```vhd :bug.vhd:

entity e is end;

use std.textio.all;

architecture a of e is
    type t_line_vec is array(natural range <>) of line;
    type t_ptr_line_vec is access t_line_vec;

    procedure p is 
        variable l : line := new string'("");
        variable lv : t_line_vec(0 to 0) := (0 => l);
        variable plv : t_ptr_line_vec := new t_line_vec'(lv);

        
        variable plv2 : t_ptr_line_vec(plv'range); --suspicious variable
    begin
        report integer'image(plv2'length); -- No crash if this line deleted. No crash if plv'length is reported instead.
    end;
begin
    process
    begin
        report "top";
        p;
        report "bottom";
        assert false severity failure;
    end process;
end;

```
Invocation: 
```powershell
C:\Users\u\Desktop\bug4> ghdl -a .\bug3.vhd; ghdl -r work.e
.\bug3.vhd:22:9:@0ms:(report note): top
PS C:\Users\u\Desktop\bug4> 
```

**Context**
Nightly GHDL Mcode windows, downloaded from CI source. (Also likely in GHDL mcode linux, but with different codebase and different git revision, and different error)
On windows, ghdl --version returns:
```
GHDL 3.0.0-dev (2.0.0.r527.g539adbe0f) [Dunoon edition]
 Compiled with GNAT Version: 12.1.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2022-07-28:
I think ghdl is correct: the variable plv2 is initialized to null, and 'range implies a dereference.
Ok, plv2 has a constrained subtype, so I am not completely sure.

Farmadupe, 2022-07-28:
From reading the 2008 LRM:
```
5.4.1 General
...
For each access type, there is a literal null that has a null access value designating no object at all
```

```
16.2.3 Predefined attributes of arrays
...
A'LENGTH [(N)] 
Kind: Function.
Prefix: Any prefix A that is appropriate for an array object, or an alias thereof,
or that denotes an array subtype whose index ranges are defined by a
constraint.
```

My (possibly incorrect) understanding of the LRM  is that since the 'length attribute requires the prefix to be "appropriate for an array object" and null is explicitly not any object, that GHDL is correct here, and the repro I pasted contains no bug.

I have a feeling that the original problem I experienced (compilation crash at trans-chap3.adb:3318) may have been lost while trying to bisect the error to build the repro. This is because my original code compiled and executed in modelsim, and it does not crash due to null pointer dereference.

I will close this issue now, and raise a new issue if I can successfully bisect my problem at my work environment.

closed 2022-07-28 by Farmadupe

diogratia, 2022-07-29:
IEEE Std 1076-1993 Names (also found in -2008 8.Names, 8.1 General):

>The evaluation of a name determines the named entity denoted by the name. The evaluation of a name that has a prefix includes the evaluation of the prefix, that is, of the corresponding name or function call. If the type of the prefix is an access type, the evaluation of the prefix includes the determination of the object designated by the corresponding access value. In such a case, it is an error if the value of the prefix is a null access value. It is an error if,after all type analysis (including overload resolution) the name is ambiguous.

So the expression
```vhdl
plv2'length
```
an attribute name, has a prefix that is an access type and should produce an error report and cause simulation to quit. The access value of plv2 hasn't been assigned the the value of an allocated variable object and is null. It can help to be familiar with 6.6 (8.6 -2008) Attribute names.

A recent macos mcode nightly version (ghdl-macos-12-mcode.tgz) demonstrates the error report:

```none
%: ghdl -a e.vhdl
%: ghdl -r e
e.vhdl:22:9:@0ms:(report note): top
ghdl:error: NULL access dereferenced
in process .e(a).P0
  from: [unknown caller]
  from: process work.e(a).P0 at e.vhdl:23
ghdl:error: simulation failed
%: ghdl --version
GHDL 3.0.0-dev (2.0.0.r527.g539adbe0f) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%:
```

The way to avoid the error is to evaluate the access value as non NULL before evaluating plv2 as a prefix:

```vhdl
entity e is end;

use std.textio.all;

architecture a of e is
    type t_line_vec is array(natural range <>) of line;
    type t_ptr_line_vec is access t_line_vec;

    procedure p (allocate_plv2: boolean := FALSE) is 
        variable l : line := new string'("");
        variable lv : t_line_vec(0 to 0) := (0 => l);
        variable plv : t_ptr_line_vec := new t_line_vec'(lv);

        
        variable plv2 : t_ptr_line_vec(plv'range); --  := new t_line_vec'(lv); 
    begin
        if allocate_plv2 then  -- allocate plv2
            plv2 := new t_line_vec'(0 => null); -- line element is unallocated
        end if;
        if plv2 /= null then 
            report integer'image(plv2'length); -- No crash if this line deleted. No crash if plv'length is reported instead.
        else
            report "plv2 = null";
        end if;
    end;
begin
    process
    begin
        report "top";
        p;
        report "between";
        p (allocate_plv2 => TRUE);
        report "bottom";
        -- assert false severity failure;
        wait;
    end process;
end;
```