ghdl/ghdl #2154: Crash on every testbench with mcode backend

bradleyharden, 2022-07-31:
**Description**
With a few minor adjustments, GHDL was finally able to analyze all of my existing VHDL-2008 code. Unfortunately, every single VUnit test crashes with the same error.

**Expected behaviour**
Not crash

**How to reproduce?**
I can't really minimize this, because it happens with every test bench. Let me know if you have suggestions for narrowing this down.

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Podman container running `ghdl/vunit:mcode-master` with `GHDL 3.0.0-dev (v2.0.0-555-g448932967)`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (v2.0.0-555-g448932967) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/project/
Command line:
/usr/local/bin/ghdl --elab-run --std=08 --work=... --workdir=... -P/project/vunit_out/ghdl/libraries/vunit_lib -P/project/vunit_out/ghdl/libraries/osvvm -P... -frelaxed ... tb -grunner_cfg=... --assert-level=error
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : ortho_code-exprs.adb:67 index check failed
Call stack traceback locations:
0x55834b48dce1 0x55834b490426 0x55834b4a8513 0x55834b7c4a89 0x55834b789606 0x55834b789b7f 0x55834b78a0bd 0x55834b7897a5 0x55834b78b8eb 0x55834b78b9a4 0x55834b78eb97 0x55834b78ee1b 0x55834b78ec28 0x55834b746260 0x55834b759974 0x55834b759b41 0x55834b7467f0 0x55834b746859 0x55834b759983 0x55834b759b79 0x55834b7caa49 0x55834b7790b1 0x55834b7cbbdb 0x55834b73713f 0x55834b7348ee 0x55834b742ac6 0x55834b7d7fe7 0x55834b712acb 0x55834b623693 0x55834b7db957 0x55834b3b9d1d 0x7fdb152ffd08 0x55834b3b82a8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2022-08-01:
Please, provide a *standalone* reproducer (file all the necessary units, even those from vunit)

bradleyharden, 2022-08-09:
Closing as a likely duplicate of #2055.

closed 2022-08-09 by bradleyharden