ghdl/ghdl #2155: Can't cast between closely related array types in VHDL-2008

bradleyharden, 2022-07-31:
**Description**
I believe the LRM was updated in 2008 to allow arrays to be closely related if their element types are closely related. However, this does not seem be accepted by GHDL.

**Expected behaviour**
If I define two array types with closely related elements, I should be able to cast between them without iterating over their elements.

**How to reproduce?**

```vhd :file: closely_related_arrays.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package closely_related_arrays is

  type sulv_vector is array (natural range <>) of std_ulogic_vector;
  type unsigned_vector is array (natural range <>) of u_unsigned;

  constant sv : sulv_vector(1 downto 0)(7 downto 0) := (1 => x"FF", 0 => x"00");
  constant uv : unsigned_vector(sv'range)(sv'element'range) := unsigned_vector(sv);

end package;
```

```sh :image: ghdl/ghdl:ubuntu22-mcode
ghdl -a --std=08 closely_related_arrays.vhd
```
```
closely_related_arrays.vhd:11:79:error: conversion allowed only between closely related types
  constant uv : unsigned_vector(sv'range)(sv'element'range) := unsigned_vector(sv);
                                                                              ^
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Podman running `ghdl/ghdl:ubuntu22-mcode` with `GHDL 3.0.0-dev (v2.0.0-527-g539adbe0f)`

tgingold, 2022-08-01:
I don't think they are closely related.
unsigned_vector and sulv_vector are two arrays whose element type is different.

bradleyharden, 2022-08-01:
@tgingold, see [this post](https://stackoverflow.com/questions/25906931/shouldnt-these-types-be-closely-related) covering the relevant change in the LRM from 93 to 2008.

commit ba2db58f (2022-08-07): vhdl-sem_names.adb(are_types_closely_related): handle vhdl08 definition

Fix #2155
closed 2022-08-07 by tgingold