ghdl/ghdl #2156: Ambiguous overloading of `to_string` for user-defined types

bradleyharden, 2022-07-31:
**Description**
Defining a `to_string` function with default arguments for a user-defined type ends up creating an ambiguity with some function defined internally to GHDL. To prevent the error, I have to add and extra argument to disambiguate the call.

**Expected behaviour**
I'm not sure what the LRM says about this, but in Questasim, I don't get any conflict with overloading.

**How to reproduce?**
Suppose you have a custom `frequency` type defined in a package, along with a `to_string` function for printing. The `to_string` function takes 3 arguments, but the latter 2 have default arguments.

```vhd :file: timing_pkg.vhd
package timing_pkg is

  type frequency is range 0.0 to real'high;

  function to_string(
    freq : frequency;
    unit : string := "";
    fmt  : string := "%.6f")
    return string;

end package;

package body timing_pkg is

  function to_string(
    freq : frequency;
    unit : string := "";
    fmt  : string := "%.6f")
    return string
  is
  begin
    if unit = "" then
      if freq < 1.0e3 then
        return to_string(real(freq), fmt) & " Hz";
      elsif 1.0e3 <= freq and freq < 1.0e6 then
        return to_string(real(freq) * 1.0e-3, fmt) & " kHz";
      elsif 1.0e6 <= freq and freq < 1.0e9 then
        return to_string(real(freq) * 1.0e-6, fmt) & " MHz";
      else
        return to_string(real(freq) * 1.0e-9, fmt) & " GHz";
      end if;
    elsif unit = "Hz" then
      return to_string(real(freq), fmt) & " Hz";
    elsif unit = "kHz" then
      return to_string(real(freq) * 1.0e-3, fmt) & " kHz";
    elsif unit = "MHz" then
      return to_string(real(freq) * 1.0e-6, fmt) & " MHz";
    elsif unit = "GHz" then
      return to_string(real(freq) * 1.0e-9, fmt) & " GHz";
    else
      report "invalid frequency unit (Hz, kHz, MHz, GHz)"
      severity failure;
    end if;
  end function;

end package body;
```

Calling this function is apparently ambiguous when only using 1 argument, as the following test shows. But if I add a dummy argument, the call is disambiguated, and everything works as expected.

```vhd :file: tb_to_string_overloading.vhd
use work.timing_pkg.all;

entity tb_to_string_overloading is
end entity;

architecture tb of tb_to_string_overloading is
  constant freq : frequency := 40.0e6;
begin

  process
  begin
    --report to_string(freq, "");
    report to_string(freq);
    wait;
  end process;

end architecture;
```

```sh :image: ghdl/ghdl:ubuntu22-mcode
ghdl -a --std=08 timing_pkg.vhd tb_to_string_overloading.vhd
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Podman running `ghdl/ghdl:ubuntu22-mcode` with `GHDL 3.0.0-dev (v2.0.0-527-g539adbe0f)`

Paebbels, 2022-07-31:
I think what you're looking for is [`-fexplicit`](https://ghdl.github.io/ghdl/using/InvokingGHDL.html?highlight=fexplicit#cmdoption-ghdl-fexplicit).

How can this function collide?  
`frequency` is a new real type, so only default functions and operators are created by the language.  
There is no `to_string` created for new real types.

----------
Btw. you should define frequency as a new physical type rather then real type.

bradleyharden, 2022-07-31:
> Btw. you should define frequency as a new physical type rather then real type.

I think that was my original approach, but I think only integer types are allowed, right? That didn't work for my use case.

Paebbels, 2022-07-31:
Yes, physical types are internally integer based, but not limited to the range of THE `integer` :).

E.g. in Vivado all pyhsical types are 64 bits (despite the fact that the bounds should be limited by universal integer (which seems to be here 64-bits), but then universal integers first incarnation called `integer` should have this range, but it's just 32-bit ...).

Anyhow, I have a physical library for it working fine from `Hz` to `GHz`.  
See [physical.vhdl](https://github.com/VLSI-EDA/PoC/blob/master/src/common/physical.vhdl?ts=2)

bradleyharden, 2022-07-31:
@Paebbels, I also remember running into problems with physical types and synthesis. Really basic stuff was not supported when computing constants with them. That's part of the reason I fell back to a `real`.

Paebbels, 2022-07-31:
For synthesis, it was tested with Quartus and ISE 14.7 and Vivado (fixed since 2019.1) back in 2015.

bradleyharden, 2022-07-31:
We use Synplify Pro for Actel/Microsemi/Microchip FPGAs. They claim to support physical types, but I ran into a mountain of bugs.

bradleyharden, 2022-07-31:
The documentation recommends against using `-fexplicit`. It seems like I should avoid it and manually disambiguate. But why do I need to do that for GHDL and not Questasim? Is that just a non-conformance to the LRM?

Paebbels, 2022-07-31:
Does GHDL report where the other overload is from. As said before, there should exist any `to_string` with same signature.

bradleyharden, 2022-07-31:
@Paebbels, I forgot to include the error message. I just added it. Yes, it reports that the other `to_string` comes from the type itself.

bradleyharden, 2022-07-31:
Ok, after some further discussion on Matrix, I think we can conclude that this is a use case for `-fexplicit` or a manual alias. I'll close.

diogratia, 2022-08-01:
-2008 also allows physical types to be defined:in a range wider than type INTEGER as Patrick commented. 5.2.4 Physical types,5.2.4.1 General:

>An implementation may restrict the bounds of the range constraint of a physical type. However, an
implementation shall allow the declaration of any physical type whose range is wholly contained within the
bounds –2147483647 and +2147483647 inclusive.

which would allow the use of smaller units (e.g. millihertz, centihertz, decihertz) to provide accuracy potentially lost using a universal_integer based range..

5.2.6 Predefined operations on scalar types

> Given a type declaration that declares a scalar type T, the following operations are implicitly declared
immediately following the type declaration (except for the TO_STRING operations in package
STANDARD, which are implicitly declared at the end of the package declaration):
>  
 >**function** MINIMUM (L, R: T) **return** T;
>  
 >**function**( MAXIMUM (L, R: T) **return** T;
>  
>**function** TO_STRING(VALUE: T) return STRING;
> ...  
>The TO_STRING operation returns the string representation (see 5.7) of the value of its actual parameter.
The result type of the operation is the type STRING defined in package STANDARD.
> ...

The TO_STRING function is implicitly declared immediately following the type declaration in package timing_pkg.

Consider:

```vhdl
package timing_pkg is

  type frequency is range 0.0 to real'high;

  function to_string(
    freq : frequency;
    unit : string := "";
    fmt  : string := "%.6f")
    return string;

  function to_string(value: frequency) return string;
end package;

package body timing_pkg is

  function to_string(
    freq : frequency;
    unit : string := "";
    fmt  : string := "%.6f")
    return string
  is
  begin
    if unit = "" then
      if freq < 1.0e3 then
        return to_string(real(freq), fmt) & " Hz";
      elsif 1.0e3 <= freq and freq < 1.0e6 then
        return to_string(real(freq) * 1.0e-3, fmt) & " kHz";
      elsif 1.0e6 <= freq and freq < 1.0e9 then
        return to_string(real(freq) * 1.0e-6, fmt) & " MHz";
      else
        return to_string(real(freq) * 1.0e-9, fmt) & " GHz";
      end if;
    elsif unit = "Hz" then
      return to_string(real(freq), fmt) & " Hz";
    elsif unit = "kHz" then
      return to_string(real(freq) * 1.0e-3, fmt) & " kHz";
    elsif unit = "MHz" then
      return to_string(real(freq) * 1.0e-6, fmt) & " MHz";
    elsif unit = "GHz" then
      return to_string(real(freq) * 1.0e-9, fmt) & " GHz";
    else
[... 50 lines cut ...]
%: ghdl -a --std=08 timing_package.vhdl
timing_package.vhdl:62:14:error: no declaration for "to_string_options"
      return to_string_options(input, "", "%.6f");
             ^
timing_package.vhdl:68:21:error: can't resolve overload for function call, slice or indexed name
    report to_string(freq);
                    ^
timing_package.vhdl:68:21:error: possible interpretations are:
timing_package.vhdl:60:12:error: function to_string [frequency return string]
timing_package.vhdl:5:12:error: function to_string [frequency, string, string return string]
ghdl:error: compilation error
```
and represents a ghdl bug.

The intent of the standard is that you can hide the implicit declaration with an explicit delaration. This represents a missing visibility check for the -2008 implicit declarations:

-1993 through -2002)  10.3 Visibility

>Two declarations that occur immediately within the same declarative region must not be homographs, unless exactly one of them is the implicit declaration of a predefined operation. In such cases, a predefined operation is always hidden by the other homograph. Where hidden in this manner, an implicit declaration is hidden within the entire scope of the other declaration (regardless of which declaration occurs first); the implicit declaration is visible neither by selection nor directly. 

The only predefined operations other than operators that come to mind for previous revisions of the standard are deallocate and file operations. This issue may be restricted to 5.2.6 Predefined operations on scalar types.

This issue should be re-opened or a new issue should be opened (@bradleyharden, @umarcor, @Paebbels, I can't reopen this issue as would be my preference).

This represents a VHDL visibility rule unimplemented by ghdl that may be restricted to a -2008 feature.

Paebbels, 2022-08-01:
@diogratia The sentence about `to_string` being implicitly declared reads a little bit as if it only applies to `std.standard` but not to types being defined in other packages / user-defined types.

Do you thing the wording is good?

I still would say, the intent of the LRM is to give each user-defined type automatically at least on `to_string` implementation. Further variants can be written by the user themselves.

diogratia, 2022-08-01:
@Paebbels I don't find

>Given a type declaration that declares a scalar type T, the following operations are implicitly declared
immediately following the type declaration (except for the TO_STRING operations in package STANDARD, which are implicitly declared at the end of the package declaration): ...

ambiguous I think there may be one too many commas, and the other one of them may be out of place. Type STRING is defined half way through package STANDARD which precludes declaring TO_STRING subprograms prior.

tgingold, 2022-08-05:
I am closing this issue as there is no bug.  I will add a test for the issue.

closed 2022-08-05 by tgingold