ghdl/ghdl #2157: PSL: next_event_a operator misbehavior

tmeissner, 2022-08-01:
**Description**
This is a tricky one ... I've stumbled over some misbehaviour of the PSL `next_event_a` operator. The form `next event a(c)[i to j](f)` indicates that `f` should hold on all of the `i`'th through `j`'th occurrences of `c`. However, the implementation in GHDL only works partly in that way.

```shell
a: _-____________
b: 44333447744433
c: _____-___---__
```
```vhdl
assert always ((a and b = x"4") -> next_event_a(c)[1 to 4](b = x"4"))
```
should hold, but fails.

```shell
a: __-_______
b: 3355555555
c: ____--_--_
```
```vhdl
assert always ((a and b = x"5") -> next_event_a(c)[1 to 4](b = x"5"))
```
holds.

Comparing the two examples, GHDL seems to have a problem with conditions other than the given one (`b = x"4"` in the 1st example), even if the "enable" condition (`c` in the 1st example) isn't true. If the values of b between the first and second occurrence of c are set to `x"4"`, the 1st example also holds.

**Expected behaviour**
Both assertions should hold. The examples are based on some on page 13 in the PSL book "A practical introduction to PSL".

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_21xx.vhd

```sh :image: ghdl/vunit:llvm
ghdl -a --std=08 issue_21xx.vhd
ghdl -e --std=08 test_issue
```

If you run it:

```sh :image: hdlc/formal:all
$ ghdl -r --std=08 test_issue
issue_21xx.vhd:137:3:@8ns:(psl assertion error): NEXT_EVENT_0_a failed
issue_21xx.vhd:137:3:@9ns:(psl assertion error): NEXT_EVENT_0_a failed
simulation stopped @30ns with status 
```
The assertion should hold.

- OS: Debian Bullseye
- Origin:
  - [x] Docker (Image:hdlc/formal:all)

tmeissner, 2022-08-01:
I assume that this is also a case for fixing something in the PSL rewriting logic. I'm currently reading into the code, maybe I have a possible fix (if I understand the code with my limited knowledge 😉)

closed 2022-08-04 by tgingold

tmeissner, 2022-08-04:
Thanks for fixing this 🙂 Now all PSL constructs under investigation in my [psl_with_ghdl repo](https://github.com/tmeissner/psl_with_ghdl#supported-but-under-investigation) are behave as they should 😀 (in simulation, at least). Have to try with synthesis later.

Liveness proofs with `eventually!` are another story, as it's more a Yosys thing, I assume.

tmeissner, 2022-08-04:
I can confirm that synthesis (and formal verification with Yosys) also work correctly now.

tgingold, 2022-08-05:
I was not worried about synthesis, as it mostly uses the same code as simulation.
So what is the next step ?  Implement stable(), prev(), rose() for simulation ?

tmeissner, 2022-08-05:
> So what is the next step ? Implement stable(), prev(), rose() for simulation ?

Yes, IMHO it would be good to try to support constructs like PSL functions in both, simulation & synthesis. Moreover there were some questions or issues from users which wanted use the functions in simulation and were confused. For reference: #662 #1498 #1530

commit 7edca407 (2022-08-04): psl-build.adb: disable incorrect optimization.  Fix #2157