ghdl/ghdl #2160: Generic parameters are not treated as locally constant

bjourne, 2022-08-04:
MWE compiled with GHDL 1.0.0, GNAT 12.1.0:

    entity systolic is
        generic(
            N : integer
        );
    end entity;
    architecture rtl of systolic is
        constant LAST : integer := 2*N - 1;
    begin
        process
            variable cnt : natural;
        begin
            case cnt is
                when LAST =>
                    report "ok";
                when others =>
                    report "ok";
            end case;
        end process;
    end rtl;

    ghdl -a --std=08 cases.vhdl                          
    cases.vhdl:13:13:error: choice is not locally static
    ghdl:error: compilation error

Shouldn't this work since generic parameters are constants after instantiation?

Paebbels, 2022-08-04:
I think GHDL is correct (doesn't mean the language couldn't be relaxed in a new revision ...)

The case statement is checked if all cases are complete and that there's no overlap. This check can't be done at analysis time, because `LAST` is unknown.

bjourne, 2022-08-04:
So the entity is not instantiated until after analysis? Is there a workaround?

tmeissner, 2022-08-04:
Using an if statement? 😉

Paebbels, 2022-08-05:
If statements and/or lookup array + for-loop might be another solution.

The same happens with constants from packages.

bjourne, 2022-08-05:
Will it synthesize to efficient rtl?

tgingold, 2022-08-07:
I do think so.  For example Yosys rewrites the rtl during optimization because it uses a different representation.

closed 2022-08-07 by tgingold