ghdl/ghdl #2161: some linux related bug

venkateshsathar, 2022-08-04:
**Description**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/spire/workspace/fpga-dev-framework2/fpga/top-level/saam/common/adsb_ais/hdl/test/
Command line:
/usr/bin/ghdl-mcode -e --workdir=workdir_TB_AD7381RegController --ieee=synopsys TB_AD7381RegisterController
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : files_map.adb:83
Call stack traceback locations:
0x56493ec38491 0x56493ec3aa53 0x56493ec44b2d 0x56493ecb6174 0x56493ecb6c85 0x56493ecb6d72 0x56493ed8b55a 0x56493ed8b283 0x56493ed8b391 0x56493ed8cb45 0x56493ed9db7a 0x56493ee55533 0x56493ed9f4f0 0x56493ed32557 0x56493ee58f31 0x56493eb3f5ff 0x7f915df64081 0x56493eb3e12c 0xfffffffffffffffe
******************************************************************
make: *** [Makefile:55: TB_AD7381RegisterController.vcd] Error 2
```
**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2022-08-05:
I need a reproducer for this issue.

Paebbels, 2022-08-05:
GHDL 0.37 is very old. Can you try with a latest version?

closed 2022-08-07 by tgingold