ghdl/ghdl #2169: Unsigned multiplication is not working as expected

bewimm, 2022-08-09:
**Description**
This started out as a synth bug, but it looks like this is also present when simulating.

bug.vhd:
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity bug is
	port(
		a :  in unsigned( 7 downto 0);
		b :  in unsigned(15 downto 0);
		r : out unsigned(23 downto 0)
	);
end entity;

architecture rtl of bug is

begin
	r <= resize(a*b, 24);
end architecture;
```

tb.vhd
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity tb_mul is
end entity;

architecture tb of tb_mul is
	signal a : unsigned( 7 downto 0);
	signal b : unsigned(15 downto 0);
	signal r : unsigned(23 downto 0);
begin
	
	u0 : entity work.bug
	port map(
		a => a,
		b => b,
		r => r
	);
	
	process
	begin
		a <= to_unsigned(243,8);
		b <= to_unsigned(34560,16);
		wait for 1 ns;
		report integer'image(to_integer(r));
		assert r = to_unsigned(8398080, 24);
		wait;
	end process;

end architecture;
```

**Expected behaviour**
I expect the assertion to not fail:
```
> ghdl -a bug.vhd
> ghdl -a tb.vhd
> ghdl -r tb_mul
tb.vhd:26:17:@1ns:(report note): 9472
tb.vhd:27:17:@1ns:(assertion error): Assertion violation
```
If I synthesize the design first then I get the same error but the output gives a clue as to what might be going wrong:
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
entity bug is
  port (
    a: in unsigned (7 downto 0);
    b: in unsigned (15 downto 0);
    r: out unsigned (23 downto 0)
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of bug is
  subtype typwrap_a is std_logic_vector (7 downto 0);
  signal wrap_a: typwrap_a;
  subtype typwrap_b is std_logic_vector (15 downto 0);
  signal wrap_b: typwrap_b;
  subtype typwrap_r is std_logic_vector (23 downto 0);
  signal wrap_r: typwrap_r;
  signal n1_o : std_logic_vector (23 downto 0);
  signal n2_o : std_logic_vector (23 downto 0);
  signal n3_o : std_logic_vector (23 downto 0);
begin
  wrap_a <= typwrap_a(a);
  wrap_b <= typwrap_b(b);
  r <= unsigned(wrap_r);
  wrap_r <= n3_o;
  -- bug.vhd:16:22
  n1_o <= "0000000000000000" & wrap_a;  --  uext
  -- bug.vhd:16:22
  n2_o <= "00000000" & wrap_b;  --  uext
  -- bug.vhd:16:22
  n3_o <= std_logic_vector (resize (signed (n1_o) * signed (n2_o), 24));
end rtl;
```
the ```resize``` puts the sign bit ('0') in the MSB instead of the correct value for unsigned.
If I synthesize as verilog then the output is working as expected.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `62a87242d`

tgingold, 2022-08-09:
I have fixed the synth issue.
But I was not able to reproduce the simulation issue.  Which backend are you using ?  What is the output of `ghdl --version` ?

bewimm, 2022-08-09:
I'm using the gcc backend:
```
GHDL 3.0.0-dev (v2.0.0-579-g62a87242d) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

commit 9034e7fb (2022-08-10): synth-vhdl_oper.adb: fix mul uns uns.  Fix #2169
closed 2022-08-10 by tgingold