ghdl/ghdl #2170: test failures on armhf in Debian and Ubuntu builds

enr0n, 2022-08-09:
**Description**
I am seeing some test failures in Debian and Ubuntu package builds, only for the armhf architecure. I suspect this is because armhf is 32-bit. Specifically, `gna/bug071` fails with the following:

```
gna bug071: [31mfailed[0m
analyze atod.vhdl
elaborate and simulate atod
./atod:error: NULL access dereferenced
in process .atod(behav).P0
./atod:error: simulation failed
```
See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1010539 for more details.

Additionally, after patching that test out, I hit this:

```
gna issue1389: [31mfailed[0m
analyze mwe.vhdl
elaborate and simulate mwe
./mwe:error: index (15232) out of bounds (0 to 128) at ../../src/ieee/math_real-body.vhdl:1163
in process .mwe(a).P0
./mwe:error: error during elaboration
```
There may be other failures too.

**Expected behaviour**
Tests should pass on armhf, or not run if they are expected to fail.

**How to reproduce?**
I believe it should be sufficient to run the test suite on 32-bit arm (maybe any 32-bit arch?).

**Context**

- OS: Ubuntu 22.10 (development branch)
- Origin: https://launchpad.net/ubuntu/+source/ghdl/1.0.0+dfsg-8

umarcor, 2022-08-10:
@enr0n are you using either LLVM or GCC? The mcode backend is not supported on ARM (neither 32 nor 64). For now, mcode is for x86 only (either 32 or 64 on GNU/Linux, 32 only on Windows).

enr0n, 2022-08-10:
Yes, this is with the LLVM backend.

tgingold, 2022-10-11:
We do test ghdl on a 32b architecture (ok, x86).
How can I investigate the failure on armhf when I don't own such a computer ?

umarcor, 2022-10-18:
@tgingold you can load qemu-user-static into memory using the `-p` (persistent) option. Then, execute "foreign" containers (e.g. `arm32v7/ubuntu:22.04`) on your x86 machine (or on GitHub Actions). Actually, this functionality is provided by default on Docker for Windows. You can find info about this topic in [dbhi.github.io/qus](https://dbhi.github.io/qus/) (disclaimer: I maintain dbhi/qus).

By using the Action from dbhi/qus on GitHub Actions, you can do:

```yml
  - uses: actions/checkout@v2
  
  - uses: dbhi/qus/action@main
    with:
      targets: arm aarch64
      
  - run: |
      docker run --rm -iv $(pwd):/wrk -w /wrk arm32v7/ubuntu:22.04 bash <<'EOF'
          apt update -qq
          apt update -y ...
          configure
          make
          make install
          run testsuite
        EOF
```

Locally, you can use the following command instead of the Action: `docker run --rm --privileged aptman/qus -s -- -p arm`. That will load and register the qemu-user-static interpreter, and it will work until you restart the machine.

Naturally, you can also install and run qemu-user-static using the package manager on your system. dbhi/qus is nothing else than putting upstream QEMU binaries in containers to make distribution independent of the host system.