ghdl/ghdl #2175: No element bounds for self-determined aggregate

bradleyharden, 2022-08-11:
**Description**
I get an error for "no element bounds for self-determined aggregate". This code compiled successfully with GHDL in the past, so I'm not sure what changed. However, while trying to find a workaround, I ran into a crash.

**Expected behaviour**
I expected this code to compile, and I expected GHDL not to crash when trying to find a workaround.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;

package pkg is

  type sulv_vector is array (natural range <>) of std_ulogic_vector;

  subtype NULL_RANGE is natural range 0 downto 1;
  constant NULL_SULV : std_ulogic_vector(NULL_RANGE) := (others => '0');
  constant NULL_SULV_VECTOR : sulv_vector(NULL_RANGE)(NULL_RANGE) := (others => NULL_SULV);

  function repeat(
    val : std_ulogic;
    n   : natural
  ) return std_ulogic_vector;

  function repeat(
    val : std_ulogic_vector;
    m   : natural
  ) return sulv_vector;

  function align_left(
    vec : std_ulogic_vector;
    len : positive;
    pad : std_ulogic := '0'
  ) return std_ulogic_vector;

end package;

package body pkg is

  function repeat(
    val : std_ulogic;
    n   : natural)
    return std_ulogic_vector
  is
  begin
    if n = 0 then
      return NULL_SULV;
    else
      return (n - 1 downto 0 => val);
    end if;
  end function;

  function repeat(
    val : std_ulogic_vector;
    m   : natural
  ) return sulv_vector
  is
  begin
    if m = 0 then
      return NULL_SULV_VECTOR;
    else
      return (m - 1 downto 0 => val);
    end if;
  end function;

--  function repeat(
--    val : std_ulogic_vector;
--    m   : natural
--  ) return sulv_vector
--  is
--    constant result : sulv_vector(m downto 1)(val'range) := (others => val);
--  begin
--    return result;
--  end function;

  function align_left(
    vec : std_ulogic_vector;
    len : positive;
    pad : std_ulogic := '0')
    return std_ulogic_vector
  is
    constant diff : integer := len - vec'length;
    alias v : std_ulogic_vector(vec'length - 1 downto 0) is vec;
    variable result : std_ulogic_vector(len - 1 downto 0);
  begin
    assert diff >= 0
    report "align_left: cannot align larger vector into smaller vector"
    severity failure;
    result := (v, repeat(pad, diff));
    return result;
  end function;

end package body;
```

```sh :image: ghdl/ghdl:bullseye-llvm-9
ghdl -a --std=08 test.vhd
```

If you swap the commented implementation for the existing implementation, you get a crash:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (v2.0.0-594-gd4d4d0706) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/project/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -fpic -c -o test.o test.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap7.adb:3218 range check failed
Call stack traceback locations:
0x6c42ce 0x6c3c99 0x6c6600 0x6c66bc 0x677b87 0x68ab8e 0x68acfd 0x70dacb 0x6afc4c 0x70ecd1 0x6632b1 0x71d952 0x71f363 0x7209ad 0x40ca07 0x7fb5a26cad08 0x40b748 0xfffffffffffffffe
******************************************************************
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Podman running `ghdl/ghdl:bullseye-llvm-9` with `GHDL 3.0.0-dev (v2.0.0-594-gd4d4d0706)`

tgingold, 2022-08-11:
I cannot reproduce the crash.
I will file the test.

bradleyharden, 2023-01-26:
@tgingold, did e9cde36 solve this issue? It looks like there are no more failing tests in the test suite.

Since you can't reproduce the crash, fixing the bug is probably the best we can do for now. I can try to replicate the crash again with a more recent version of GHDL.

tgingold, 2023-01-27:
Indeed, I think this is now fixed.

closed 2023-01-27 by tgingold