ghdl/ghdl #2178: PSL:

tmeissner, 2022-08-14:
**Description**
Some time ago tried named sequences with parameters again and I stumbled over the fact, that `boolean` parameters seem to be supported in some way. However, the parameters cannot be used with an expression inside the sequence. When one tries that, it leads to a GHDL error.

For example, a working sequence:
```vhdl
-- This  works, but it's not that useful as even a simple negation
-- leads to an error (see seq1)
sequence seq (boolean d0, d1) is {d0; d1};

TEST_0 : cover seq(a, b) report "TEST_0 hit";
```

a not working sequence:

```vhdl
-- This doesnt work:
-- error: PSL declaration "a" not allowed in an expression
sequence seq1 (boolean d0) is {not d0; d0 = '1'};
  
TEST_1 : cover seq1(a) report "TEST_1 hit";
```

```shell
issue_2178.vhd:75:38:error: PSL declaration "d0" not allowed in an expression
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
```

**Expected behaviour**
I would expect that all valid operations on `boolean` should be supported with `boolean` parameters inside sequences.

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_2178.vhd

```sh :image: ghdl/vunit:llvm
ghdl -a --std=08 issue_2178.vhd
ghdl -e --std=08 test_issue
```

If you run it:

```sh :image: hdlc/formal:all
$ ghdl -a --std=08 issue_2178.vhd 
issue_2178.vhd:75:38:error: PSL declaration "d0" not allowed in an expression
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
                                     ^
issue_2178.vhd:75:34:error: no function declarations for operator "not"
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
                                 ^
issue_2178.vhd:75:34:error: type of expression must be boolean
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
                                 ^
issue_2178.vhd:75:42:error: PSL declaration "d0" not allowed in an expression
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
                                         ^
issue_2178.vhd:75:45:error: no function declarations for operator "="
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
                                            ^
issue_2178.vhd:75:45:error: type of expression must be boolean
  sequence seq1 (boolean d0) is {not d0; d0 = '1'};
                                            ^
issue_2178.vhd:91:21:error: entity 'test_issue' was not analysed
architecture sim of test_issue is
                    ^
```
Most of these errors are follow-ups of the not supported expression use.

- OS: Debian Bullseye
- Origin:
  - [x] Docker (Image:hdlc/formal:all)

tgingold, 2022-08-15:
Interesting.  But I am not sure that this is correct:
>   sequence seq1 (boolean d0) is {not d0; d0 = '1'};
`d0` is a PSL boolean, so it is seen as a VHDL boolean according to ieee1850, isn't it ?

Then ieee1850 is not clear on how the conversion is done.  Ok, '1' is true, '0' is false, but what about 'X' or 'Z' ?

tmeissner, 2022-08-15:
Yeah, you're probably right. `d0 = true` would be valid, or simply `d0`.
This is an artifact of my tries to implement support of parameters of type `bit`, I assume.