ghdl/ghdl #2179: Can't use others aggregate with unconstrained elements

Paebbels, 2022-08-17:
**Description**
When a port is a record type containing unconstrained elements, and such a port's element is assigned from an aggregate using `others`, then GHDL reports an error.

> reproducer.vhdl:20:19: 'others' choice not allowed for an aggregate in this context

**Expected behavior**
When a local signal declared from an unconstrained record type is used, GHDL reports no error.  
When the aggregate uses `'range` of the assignment target, no error is reported.

I think this assignment should be valid in GHDL too.  
Vivado and Riviera-PRO don't report errors on this statement.

**How to reproduce?**
The following is a minimal example. The reported line is marked as a comment.

```vhd :file: ent.vhd
package pkg is
	type rec is record
		elem : bit_vector;
	end record;
end package;

entity i is
	port (
		signal p : out work.pkg.rec
	);
end entity;

architecture a of i is
	signal s1 : work.pkg.rec(elem(3 downto 0));
	signal s2 : s1'subtype;
begin
	s1.elem <= (others => '0');
	s2.elem <= (others => '0');
	p.elem <= (p.elem'range => '0');
--	p.elem <= (others => '0');        -- reproducer.vhdl:20:19: 'others' choice not allowed for an aggregate in this context
end architecture;

entity e is
end entity;

architecture a of e is
	signal s : work.pkg.rec(
		elem(7 downto 0)
	);
begin
	inst : entity work.i
		port map (
			p => s
		);
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run e
```

**Context**
Please, provide the following information:

- OS: Windows 11, MinGW64
- Origin:
  - [x] Built from sources: `cdcbb258b`

**Additional context**
I assume, constraints are not propagated properly, so the aggregate can't access the range of the element.  
Another idea is, GHDL is blocking this behavior due to a VHDL-93 to VHDL-08 change.

asurafire, 2022-08-26:
I still see the same error message in my simulation run.

diogratia, 2022-08-28:
If you look at commit 0db659f you'd see the OP's test program is run twice, the first time it fails, the second time when passing the command line flag -frelaxed it passes. The only change in the commit is to provide the test program as a ghdl test case.

Note the 'subtype predefined attribute inferring -2008.

IEEE Std 1076-2008, 9.3.3.3 Array aggregates

>The index range of an array aggregate that has an **others** choice shall be determinable from the context. That is, an array aggregate with an **others** choice shall appear only in one of the following contexts:
>> ...
>>e) As a value expression in an assignment statement, where the target is a declared object (or member thereof), and either the subtype of the target is a fully constrained array subtype or the target is a slice name
>> ...

Both the slice name and full constrained array subtype language elements in the assignment using an aggregate with an **others** choice are new in -2008.

6.5.6.3 Port clauses

>If a formal port is associated with an actual port, signal, or expression, then the formal port is said to be _connected_. If a formal port is instead associated with the reserved word **open**, then the formal is said to be _unconnected_. It is an error if a port of mode **in** is unconnected (see 6.5.6.3) or unassociated (see 6.5.7.3) unless its declaration includes a default expression (see 6.5.2). It is an error if a port of any mode other than **in** is unconnected or unassociated and its type is an unconstrained or partially constrained composite type. It is an error if some of the subelements of a composite formal port are connected and others are either unconnected or unassociated.

That the port p would be required to be associated could suggest additional criteria for 9.3.3.3 b) or an additional rule. In the meantime with ghdl adhering closer to the standard than other implementations the -frelaxed flag is required.

Paebbels, 2022-08-28:
@tgingold I think it would be nice to provide a little comment that this codes requires `-frelaxed`, because not every user will read the uploaded test's code in details and will notice this tiny difference - I also didn't notice it.

---------
@diogratia So we need to enhance the LRM for this case to allow the same power to `others` as `'range` currently has?

closed 2022-08-28 by Paebbels

diogratia, 2022-08-28:
@Paebbels - the issue presented here is the use of an **others** choice in an array aggregate expression  whose index constraints that can't be seen from context specifically an assignment statement. It might be worth exploring the the reasoning resulting in the use of _fully constrained array subtype_ a term of art found not quite exclusively in describing where the **others** choice is used in array aggregates  and the addition of _target is a slice name_ to the 9.3.3.3 case e). Both changes made in -2008.

A solution may be as simple as defining the term of art to accommodate the the 6.5.6.3 port limitations above. The closest place to defining the term of art _fully constrained array subtype_ is in 9.4.2.

That we've seen the three so far mentioned implementations able to accommodate this issue leaving only the two vendors with the largest market share unrepresented so far says it may not impinge on existing implementations and represents a clarification.

tgingold, 2022-10-11:
So, it is not the same issue, as the test passes.  I need a reproducer.

tgingold, 2022-10-11:
Try with `-frelaxed`

commit 0db659f0 (2022-08-19): testsuite/gna: add a test and close #2179