ghdl/ghdl #2180: ghdl-macos-10.15-llvm does not analyze file

Dotto92, 2022-08-18:
I've dowloaded and installed ghdl-macos-10.15-llvm and when I try to analyze a simple vhd file, the following message returns:

`Calimero$ ghdl -a or2.vhd
dyld: Symbol not found: __ZN4llvm15SmallVectorBaseIjE8grow_podEPvmm
  Referenced from: /Users/Calimero/Applications/ghdl-macos-10.15-llvm/bin/ghdl1-llvm
  Expected in: /usr/local/opt/llvm/lib/libLLVM.dylib
 in /Users/Calimero/Applications/ghdl-macos-10.15-llvm/bin/ghdl1-llvm
ghdl:error: exec error`

The OS used is MacOS 10.15.7 (Catalina)
If I use an older version (ghdl-0.36-macosx-llvm), it works correctly.

Regards

diogratia, 2022-08-19:
The error message is saying /usr/local/opt/llvm/lib/libLLVM.dylib is not compatible with the library ghdl was compiled against. It would be from a different and incompatible Homebrew llvm release's libLLVM.dylib. ghdl doesn't distribute the libLLVM.dyib dynamic library. 

During the build a symbol was used that isn't found in your available libLLVM.dylib.

IIRC ghdl-0.36 used statically linked LLVM libraries:

```none
%: otool -L ghdl1-llvm
ghdl1-llvm:
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
	@rpath/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
%: otool -L ghdl
ghdl:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/usr/local/gnat/lib/libgcc_s.1.dylib (compatibility version 1.0.0, current version 1.0.0)
%:  
```

More recent ghdl-llvm builds use dynamic libraries.

```none
%: ghdl --version
GHDL 3.0.0-dev (2.0.0.r607.g550c8e653) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
%: pwd
/usr/local/bin
%: otool -L ghdl1-llvm
ghdl1-llvm:
	/usr/local/opt/llvm/lib/libLLVM.dylib (compatibility version 1.0.0, current version 14.0.6)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
	/usr/local/opt/llvm/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
%: 
```
By searching you'd find I'm using a nightly build from a particular commit and have llvm-14 installed while the ghdl release was built with llvm-13.  I've found llvm-13 and llvm-14 appear compatible.

You should find that ghdl --version will return information properly without calling anything affected by the wrong dylib version dynamically linked in later ghdl1-llvm versions. The release number or rXXX.gXXXXX can tie the release to a commit in the master source tree on github, that can be used to find the version of llvm used to build a release (curently by searching for commits to ghdl/.github/workflows/Test.yml changing the Homebrew llvm vesion and comparing the date against the release/commit date).
 
The release number provided by ldd (otool -L) for libLLVM.dylib (provided by Homebrew cellar as a keg) may or may not be inconsistent across llvm major releases. They are not  for llvm.org binary releases nor Apple's (neither of which supply libLLVM.dylib).

I recently found that the Homebrew llvm-12 dylib is not compatible with the  llvm-13 dylib, while it appears the llvm-13 dylib appears is compatible with the llvm-14 dylib. There are earlier incompatibilities.

The gist is you can either try and match your Homebrew llvm release to the version used to produce the ghdl executable, build a particular commit or release from sources yourself or use an mcode version of ghdl (which may only be vulnerable to incompatibilities in the MacOS SDK environment).

Dotto92, 2022-08-19:
Thank you for your answer.
I'm not a software developer and it's a bit hard for me to understand. I used the nightly build of ghdl-llvm for MacOS-10.15 coming from the git repositories. I tried to use also ghdl-macos-11.0-llvm 2 and ghdl-macos-12-llvm, but neither of them solve the problem (the error message is always the same). Do you think that building it from source code on my own machine can solve the problem?

Dotto92, 2022-08-28:
I downloaded the source code (ghdl 2.0.0) and I tried to compile it with the following configure command:

`.././configure --with-llvm-config=/usr/local/Cellar/llvm/10.0.0_3/bin/llvm-config --disable-werror --prefix=../ghdl
Use full IEEE library
Build machine is: x86_64-apple-darwin17.7.0
Debugging is enabled with llvm 10.0.0
create pic/ subdirectory
Creating ghdl.gpr
Creating Makefile
for d in ieee/v87 ieee/v93 ieee/v08 std/v87 std/v93 std/v08 src/ieee src/ieee/v87 src/ieee/v93 src/ieee2008 src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 src/upf src/vital95 src/vital2000; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads`

But after the make command, the following error returns:

```
VER_DESC=tarball; \
	VER_REF=unknown; \
	VER_HASH=unknown; \
	if test -d .././.git && desc=`cd ../.; git describe --dirty --long`; then \
	  VER_DESC=`echo $desc | sed -e 's/\([^-]*-g\)/r\1/' -e 's/-/./g' -e 's/^v//g'`; \
	  VER_REF=`cd ../.; git rev-parse --abbrev-ref HEAD`; \
	  VER_HASH=`cd ../.; git rev-parse HEAD`; \
	fi; \
	sed \
	  -e "s#@VER@#2.0.0#" \
	  -e "s#@DESC@#${GHDL_VER_DESC:-$VER_DESC}#" \
	  -e "s#@REF@#${GHDL_VER_REF:-$VER_REF}#" \
	  -e "s#@HASH@#${GHDL_VER_HASH:-$VER_HASH}#" \
	  < .././src/version.in > version.tmp;
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gcc -c  -g -Wall -g -o grt-cstdio.o /Users/Calimero/Desktop/ghdl-2.0.0/build/.././src/grt/grt-cstdio.c
/Users/Calimero/Desktop/ghdl-2.0.0/build/.././src/grt/grt-cstdio.c:17:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.
make: *** [grt-cstdio.o] Error 1
```
What is missing?

tgingold, 2022-08-29:
stdio.h is the most common header.  Did you install Xcode ?  You can also try to configure with CC=clang

Dotto92, 2022-08-30:
After the configure command, I modified the key CC=gcc with CC=clang and the compilation starts. It stops two times. The first time the following error returns:

```
gnatbind -aI.././src -aI.././src/vhdl -aI.././src/synth -aI.././src/grt -aI.././src/psl -aI.././src/vhdl/translate -aI.././src/ghdldrv -aI.././src/ortho -aI.././src/ortho/llvm6 -aI.././src/synth -aI.././src/ghdldrv -E -x ghdl_llvm.ali
gnatlink ghdl_llvm.ali -g -Wl,-no_compact_unwind grt-cstdio.o
ld: library not found for -lSystem
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/Calimero/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [ghdl_llvm] Error 4
```

I added the  --LINK=clang to the gnatlink command:

`gnatlink --LINK=clang ghdl_llvm.ali -g -Wl,-no_compact_unwind grt-cstdio.o`

and then 

`make`

The compilation continues.

The second time the following error returns:

```
gnatbind -aI. -aO/Users/Calimero/Desktop/ghdl-2.0.0/build/pic -aI.././src -aI.././src/vhdl -aI.././src/synth -aI.././src/grt -aI.././src/psl -aI.././src/vhdl/translate -aI.././src/synth -aI.././src/vhdl/libghdl -aI.././src/ghdldrv -shared -Llibghdl_ -z -x /Users/Calimero/Desktop/ghdl-2.0.0/build/pic/libghdl.ali
gnatlink /Users/Calimero/Desktop/ghdl-2.0.0/build/pic/libghdl.ali -shared-libgcc -o lib/libghdl-2_0_0.dylib -g -g -shared -Wl,-install_name,@rpath/libghdl-2_0_0.dylib -Wl,-no_compact_unwind pic/grt-cstdio.o
ld: library not found for -lSystem
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/Calimero/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [lib/libghdl-2_0_0.dylib] Error 4
```
I ran the gnatlink command (with --LINK=clang added):

`gnatlink --LINK=clang /Users/Calimero/Desktop/ghdl-2.0.0/build/pic/libghdl.ali -shared-libgcc -o lib/libghdl-2_0_0.dylib -g -g -shared -Wl,-install_name,@rpath/libghdl-2_0_0.dylib -Wl,-no_compact_unwind pic/grt-cstdio.o`

and the compilation came to end.

After make install:

```
ghdl -v
GHDL 2.0.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: Community 2020 (20200429-84)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I tried to compile a very little VHDL design and it seems to work properly.

Many thanks for your support.

Regards

tgingold, 2022-08-31:
Great!  MacOS and/or Xcode have changed and it is possible that gnat 2020 is not fully compatible with the newest versions.