ghdl/ghdl #2181: translate_subtype_definition: cannot handle IIR_KIND_INTERFACE_TYPE_DEFINITION

m-kru, 2022-08-20:
**Description**
translate_subtype_definition: cannot handle IIR_KIND_INTERFACE_TYPE_DEFINITION

**Expected behaviour**
It should work, shouldn't it?

**How to reproduce?**

```vhd :file: duration.vhd
library ieee;
   use ieee.std_logic_1164.all;
   use ieee.numeric_std.all;

-- The Duration checker ensures that given value lasts for given
-- time on given data signal.
--
-- POLICY - check policy. Must be one of: 'equal', 'less than' or 'more than'.
-- PREFIX - an optional prefix used in report messages.
entity Duration is
   generic (
      type data_type;
      function to_string(d : data_type) return string;
      POLICY : string := "equal";
      SVRITY : SEVERITY_LEVEL := failure;
      PREFIX : string := ""
   );
   port (
      data     : in data_type;
      duration : in time;
      value    : in data_type
   );
begin

   assert POLICY = "equal" or POLICY = "less than" or POLICY = "more than"
      report "POLICY must be one of: 'equal', 'less than' or 'more than'"
      severity failure;

end entity;

architecture Check of Duration is

   impure function policy_to_str return string is
   begin
      if POLICY = "equal" then
         return "";
      end if;
      return POLICY & " ";
   end function;

begin

   process is
      variable start_time : time := 0 fs;
      variable duration_time : time;
      variable violated : boolean;
   begin
      loop
         wait until data'event;
         violated := false;

         if data = value then
            start_time := now;
         elsif data'last_value = value then
            duration_time := now - start_time;
            if POLICY = "equal" then
               if duration_time /= DURATION then violated := true; end if;
            elsif POLICY = "less than" then
               if duration_time >= DURATION then violated := true; end if;
            elsif POLICY = "more than" then
               if duration_time <= DURATION then violated := true; end if;
            end if;
            if violated then
               report PREFIX & "value 0b" & to_string(value) &
                  " lasted for " & to_string(duration_time) &
                  ", but was supposed to last for " & policy_to_str & to_string(DURATION)
                  severity SVRITY;
            end if;
         end if;
      end loop;
   end process;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 duration.vhd
```

**Context**
Please, provide the following information:

- OS: 5.15.0-46-generic #49~20.04.1-Ubuntu

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r638.gc28f780bc) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/mkru/workspace/vhdl/ghdl-bug0/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -fpic -c -o duration.o duration.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x50433e 0x70120a 0x70183a 0x6a1a78 0x6a9bf6 0x712d9d 0x660aa7 0x71ad8c 0x71c79d 0x71dde7 0x40ca45 0x7f7141ebf081 0x40b76c 0xfffffffffffffffe
...
******************************************************************
```

tgingold, 2022-08-20:
Not yet, only in synthesis.

m-kru, 2022-08-22:
How far is it on the road map?

tgingold, 2022-08-23:
Not before next year.  I think it will come with external names.