ghdl/ghdl #2182: Linker error during elaboration on MSYS2

amb5l, 2022-08-27:
I am seeing a linker error during elaboration of my testbench, as follows:
```
ghdl -e --std=08 -frelaxed np6532_functest
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLD'
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLw'
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o:ortho:(//AArdLD[.refptr.ieee__numeric_std__ELABORATED]+0x0): multiple definition of `.refptr.ieee__numeric_std__ELABORATED'; D:\devtools\msys64\mingw64\lib/ghdl\ieee\v08\numeric_std-body.o:ortho:(.rdata$.refptr.ieee__numeric_std__ELABORATED[.refptr.ieee__numeric_std__ELABORATED]+0x0): first defined here
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o:ortho:(//AArdLw[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): multiple definition of `.refptr.ieee__std_logic_1164__ELABORATED'; D:\devtools\msys64\mingw64\lib/ghdl\ieee\v08\std_logic_1164-body.o:ortho:(.rdata$.refptr.ieee__std_logic_1164__ELABORATED[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```

This may or may not be related to issue #2020.

[ghdl_linker_error.zip](https://github.com/ghdl/ghdl/files/9437863/ghdl_linker_error.zip) contains a makefile plus required source etc that will demonstrate the issue. A Xilinx Vivado 2022.1 installation is required for the library sources.

OS: Windows 11 (Version 10.0.22000 Build 22000)
MSYS2 installed from https://repo.msys2.org/distrib/x86_64/msys2-x86_64-20220603.exe
GHDL installed with `pacman -S mingw-w64-x86_64-ghdl-llvm`
GHDL version:
```
GHDL 3.0.0-dev (2.0.0.r394.g4afeced94) [Dunoon edition]
 Compiled with GNAT Version: 12.1.0
 llvm code generator
```

amb5l, 2022-08-27:
I have a MCVE. The following source and makefile, pasted below, are sufficient to show the issue. This requires a Vivado 2022.1 installation at `$(XILINX_VIVADO)`.

1) source
```
library unisim;
use unisim.vcomponents.all;
entity top is
end entity top;
architecture sim of top is
begin
end architecture sim;
```

2) makefile
```
all: elaborate

GHDL=ghdl
GHDLAOPTS=--std=08 -frelaxed -fsynopsys
GHDLEOPTS=--std=08 -frelaxed

XSRC=$(XILINX_VIVADO)/data/vhdl/src
UNISIM_SRC=\
	$(XSRC)/unisims/unisim_retarget_VCOMP.vhd \
	$(XSRC)/unisims/unisim_VPKG.vhd \
	$(XSRC)/unisims/primitive/LDCE.vhd \
	$(XSRC)/unisims/primitive/RAM64X1D.vhd

UNISIM=unisim-obj08.cf

$(UNISIM): $(UNISIM_SRC)
	$(GHDL) -a $(GHDLAOPTS) --work=unisim $?

top.o: top.vhd $(UNISIM)
	$(GHDL) -a $(GHDLAOPTS) $<

top.exe: top.o
	$(GHDL) -e $(GHDLEOPTS) top

analyse: top.o

elaborate: top.exe

clean:
	rm *.o *.cf *.lst
```

amb5l, 2022-08-27:
The problem is still present when I build from the latest source:
```
GHDL 3.0.0-dev (2.0.0.r663.geefd7020a) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm code generator

```

tgingold, 2022-08-27:
Thank you for the small reproducer.  I will have a look, but I first need to open my windows computer.

Paebbels, 2023-05-04:
@stefanunrein Please post your error message here. The listed errors look similar. Was it VCOMP or retarget VCOMP in your case?

stefanunrein, 2023-05-04:
Looks quite similar with our code. This is the error log from one test-case. Five others have exactly the same problem.

```
analyze ../tb/TC_UDP_Stack_BasicSendReceive_NRL.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=TbPkg --workdir=C:/Projekte/udp-ethernet-stack/PL/temp/GHDL/VHDL_LIBS/GHDL-4.0.0-dev/tbpkg/v08 -PC:/Tools/Precompiled/GHDL/Vivado/2022.2 -PC:/Projekte/udp-ethernet-stack/PL/temp/GHDL/VHDL_LIBS/GHDL-4.0.0-dev -fexplicit -Wno-elaboration -frelaxed -Wno-port -Wno-specs -Wno-pure -frelaxed ../../tb/TC_UDP_Stack_BasicSendReceive_NRL.vhd

TestName TC_UDP_Stack_BasicSendReceive_NRL
Simulation Start time 23:32:55
simulate TC_UDP_Stack_BasicSendReceive_NRL [generic PACKETMODE_Enable false]
ghdl --elab-run --std=08 --syn-binding -frelaxed -Wno-binding --work=TbPkg --workdir=C:/Projekte/udp-ethernet-stack/PL/temp/GHDL/VHDL_LIBS/GHDL-4.0.0-dev/tbpkg/v08 -PC:/Tools/Precompiled/GHDL/Vivado/2022.2 -PC:/Projekte/udp-ethernet-stack/PL/temp/GHDL/VHDL_LIBS/GHDL-4.0.0-dev TC_UDP_Stack_BasicSendReceive_NRL -gPACKETMODE_Enable=false
Error: C:/mingw/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/Tools/Precompiled/GHDL/Vivado/2022.2\unisim\v08\unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AA1GKk'
Error: C:/Tools/Precompiled/GHDL/Vivado/2022.2\unisim\v08\unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AA1GLR'
Error: C:/mingw/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/Tools/Precompiled/GHDL/Vivado/2022.2\unisim\v08\unisim_retarget_VCOMP.o:ortho:(//AA1GKk[.refptr.ieee__numeric_std__ELABORATED]+0x0): multiple definition of `.refptr.ieee__numeric_std__ELABORATED'; C:\mingw\usr\local\lib/ghdl\ieee\v08\numeric_std-body.o:ortho:(.rdata$.refptr.ieee__numeric_std__ELABORATED[.refptr.ieee__numeric_std__ELABORATED]+0x0): first defined here
Error: C:/mingw/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/Tools/Precompiled/GHDL/Vivado/2022.2\unisim\v08\unisim_retarget_VCOMP.o:ortho:(//AA1GLR[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): multiple definition of `.refptr.ieee__std_logic_1164__ELABORATED'; C:\mingw\usr\local\lib/ghdl\ieee\v08\std_logic_1164-body.o:ortho:(.rdata$.refptr.ieee__std_logic_1164__ELABORATED[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
Error: C:\mingw\usr\local\bin\ghdl.exe: compilation error
SimulateError: See messages above in "simulate {TC_UDP_Stack_BasicSendReceive_NRL {}}"
For tcl errorInfo, puts $::osvvm::SimulateErrorInfo
ReportError: Simulate2Html failed.  See previous messages for details

```

Paebbels, 2023-05-06:
I checked `unisim_retarget_VCOMP.o` and it really contains that symbol:

```
Patrick_Lehmann@Paebbels MINGW64 /c/Users/Patrick_Lehmann/Downloads
$ objdump -t unisim_retarget_VCOMP.o | grep numeric_std
C:\msys64\mingw64\bin\objdump.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLD'
unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLw'
[ 10](sec  6)(fl 0x00)(ty    0)(scl   3) (nx 1) 0x0000000000000000 .rdata$.refptr.ieee__numeric_std__ELABORATED
[ 12](sec  6)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x0000000000000000 .refptr.ieee__numeric_std__ELABORATED
[ 33](sec  0)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x0000000000000000 ieee__numeric_std__ELAB_BODY
[205817](sec  0)(fl 0x00)(ty    0)(scl   2) (nx 0) 0x0000000000000000 ieee__numeric_std__ELABORATED
```

Paebbels, 2023-05-06:
There is also this issue:
* #2348

Paebbels, 2023-05-06:
I asked a colleague to check the generated *.o file in Windows MinGW64 and native Linux. The symbols are only added in the MinGW64 variant. Also linking on Linux works with `ld` and `lld`.

I created this short reproducer:
```vhdl
library ieee;
use     ieee.std_logic_1164.all;

library unisim;
use     unisim.vcomponents.all;

entity e is
	port (
		Clock   : in  std_logic;
		DataIn  : in  std_logic;
		DataOut : out std_logic
	);
end entity;

architecture a of e is
begin
	flip : component FD
		port map (
			C => Clock,
			D => DataIn,
			Q => DataOut
		);
end architecture;
```

It requires these Xilinx Vivado files:
* `unisim_VPKG.vhd`
* `unisim_VCOMP.vhd`
* `primitive/FDRE.vhd`
* `retarget/FD.vhd`

Compilation of Xilinx Unisim components:
```bash
ghdl -a --std=08 -frelaxed --work=unisim unisim_VPKG.vhd
ghdl -a --std=08 -frelaxed --work=unisim unisim_VCOMP.vhd
ghdl -a --std=08 -frelaxed --work=unisim FDRE.vhd
ghdl -a --std=08 -frelaxed --work=unisim FD.vhd
```

Triggering the error:
```bash
ghdl -a --std=08 -frelaxed symbols.vhdl
ghdl -e --std=08 -frelaxed e
```

Or using the precompile scripts:
```bash
ghdl -e --std=08 -frelaxed -P/c/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2 e
```

When using the proposed `lld` linker, it works:
```bash
ghdl -e --std=08 -frelaxed -P/c/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2 --LINK=clang -Wl,-fuse-ld=lld e
```

tgingold, 2023-05-06:
I am not sure to understand the last sentence: is there a linker error during elaboration on MSYS2 when using lld ?

Paebbels, 2023-05-06:
With `lld` on MinGW64, there is no error and an *.exe file is generated.

tgingold, 2023-05-06:
So it confirms this is an issue with binutils.

Paebbels, 2023-05-06:
At least binutils and MinGW64. Can you (maybe) confirm the *.o file is correctly generated by LLVM?  
Could a 4.7 MB VHDL file trigger some problems in code generation?

The *.o file contains "more" `numeric_std` symbols on MinGW64 then on native Linux. Or is this a ELF vs. COFF problem and they need to workaround COFF short comings with additional symbols?

------------
I'll try what happens if I use the non-retarget variants of VComponents. This VHDL file is smaller.

tgingold, 2023-05-06:
I suppose there is an issue in the binutils reader when there is a large number of symbols.

Paebbels, 2023-05-06:
Btw. also `objdump` complains with a warning:
```
C:\msys64\mingw64\bin\objdump.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLD'
unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLw'
```

Is `objdump` part of binutils?

Paebbels, 2023-05-06:
> I suppose there is an issue in the binutils reader when there is a large number of symbols.

The file has "only" `205827` symbols.

tgingold, 2023-05-06:
Yes, objdump is provided by binutils.
I think llvm provides llvm-objdump.  If it is installed, it might be interesting to check its output.

Paebbels, 2023-05-06:
Output from `llvm-objdump`:
```
$ llvm-objdump -t unisim_retarget_VCOMP.o | grep numeric_std
[10](sec  6)(fl 0x00)(ty   0)(scl   3) (nx 1) 0x00000000 .rdata$.refptr.ieee__numeric_std__ELABORATED
[12](sec  6)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 .refptr.ieee__numeric_std__ELABORATED
[33](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__numeric_std__ELAB_BODY
[205817](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__numeric_std__ELABORATED
```

```
$ llvm-objdump -t unisim_retarget_VCOMP.o | grep std_logic
[13](sec  7)(fl 0x00)(ty   0)(scl   3) (nx 1) 0x00000000 .rdata$.refptr.ieee__std_logic_1164__ELABORATED
[15](sec  7)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 .refptr.ieee__std_logic_1164__ELABORATED
[32](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__std_logic_1164__ELAB_BODY
[39](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__std_logic_1164__resolved_RESOLV
[2611](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__std_logic_1164__std_logic__RTI
[2654](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__std_logic_1164__std_ulogic__RTI
[136247](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__std_logic_1164__std_ulogic_vector__RTI
[205818](sec  0)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00000000 ieee__std_logic_1164__ELABORATED
```

tgingold, 2023-05-06:
So the bug is within the binutils reader.

Paebbels, 2023-05-06:
Differences in sections:

`objdump`:
```
$ objdump --section-headers unisim_retarget_VCOMP.o
C:\msys64\mingw64\bin\objdump.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLD'
unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLw'

unisim_retarget_VCOMP.o:     file format pe-x86-64

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         00007f83  0000000000000000  0000000000000000  00000244  2**4
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .data         00000000  0000000000000000  0000000000000000  00011a5b  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00002da3  0000000000000000  0000000000000000  00000000  2**4
                  ALLOC
  3 .xdata        00000010  0000000000000000  0000000000000000  00011a5b  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .rdata        0072b3e1  0000000000000000  0000000000000000  00011a6b  2**4
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA
  5 //AArdLD      00000008  0000000000000000  0000000000000000  009d2918  2**3
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA, LINK_ONCE_DISCARD (COMDAT .refptr.ieee__numeric_std__ELABORATED 12)
  6 //AArdLw      00000008  0000000000000000  0000000000000000  009d292a  2**3
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA, LINK_ONCE_DISCARD (COMDAT .refptr.ieee__std_logic_1164__ELABORATED 15)
  7 .debug_abbrev 00000028  0000000000000000  0000000000000000  009d293c  2**0
                  CONTENTS, READONLY, DEBUGGING
  8 .debug_info   0000003f  0000000000000000  0000000000000000  009d2964  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
  9 .debug_str    0000009a  0000000000000000  0000000000000000  009d29f3  2**0
                  CONTENTS, READONLY, DEBUGGING
 10 .debug_pubnames 00000035  0000000000000000  0000000000000000  009d2a8d  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
 11 .debug_pubtypes 00000012  0000000000000000  0000000000000000  009d2acc  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
 12 .pdata        0000000c  0000000000000000  0000000000000000  009d2ae8  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA
 13 .debug_line   000000db  0000000000000000  0000000000000000  009d2b12  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING

```

`llvm-objdump`:
```
$ llvm-objdump --section-headers unisim_retarget_VCOMP.o

unisim_retarget_VCOMP.o:        file format coff-x86-64

Sections:
Idx Name                                            Size     VMA              Type
  0 .text                                           00007f83 0000000000000000 TEXT
  1 .data                                           00000000 0000000000000000 DATA
  2 .bss                                            00002da3 0000000000000000 BSS
  3 .xdata                                          00000010 0000000000000000 DATA
  4 .rdata                                          0072b3e1 0000000000000000 DATA
  5 .rdata$.refptr.ieee__numeric_std__ELABORATED    00000008 0000000000000000 DATA
  6 .rdata$.refptr.ieee__std_logic_1164__ELABORATED 00000008 0000000000000000 DATA
  7 .debug_abbrev                                   00000028 0000000000000000 DATA, DEBUG
  8 .debug_info                                     0000003f 0000000000000000 DATA, DEBUG
  9 .debug_str                                      0000009a 0000000000000000 DATA, DEBUG
 10 .debug_pubnames                                 00000035 0000000000000000 DATA, DEBUG
 11 .debug_pubtypes                                 00000012 0000000000000000 DATA, DEBUG
 12 .pdata                                          0000000c 0000000000000000 DATA
 13 .debug_line                                     000000db 0000000000000000 DATA, DEBUG
```

Paebbels, 2023-05-06:
When analyzing the smaller `unisim_VCOMP.vhd` I get this:

```
$ objdump --section-headers /c/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2/unisim/v08/unisim_VCOMP.o
C:\msys64\mingw64\bin\objdump.exe: C:/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2/unisim/v08/unisim_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AAxHkl'
C:/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2/unisim/v08/unisim_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AAxHlS'

C:/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2/unisim/v08/unisim_VCOMP.o:     file format pe-x86-64

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         00002f6f  0000000000000000  0000000000000000  00000244  2**4
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .data         00000000  0000000000000000  0000000000000000  00006a2f  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          000010cf  0000000000000000  0000000000000000  00000000  2**4
                  ALLOC
  3 .xdata        00000010  0000000000000000  0000000000000000  00006a2f  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .rdata        00821ebd  0000000000000000  0000000000000000  00006a3f  2**4
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA
  5 //AAxHkl      00000008  0000000000000000  0000000000000000  00b02992  2**3
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA, LINK_ONCE_DISCARD (COMDAT .refptr.ieee__numeric_std__ELABORATED 12)
  6 //AAxHlS      00000008  0000000000000000  0000000000000000  00b029a4  2**3
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA, LINK_ONCE_DISCARD (COMDAT .refptr.ieee__std_logic_1164__ELABORATED 15)
  7 .debug_abbrev 00000028  0000000000000000  0000000000000000  00b029b6  2**0
                  CONTENTS, READONLY, DEBUGGING
  8 .debug_info   0000003f  0000000000000000  0000000000000000  00b029de  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
  9 .debug_str    0000009b  0000000000000000  0000000000000000  00b02a6d  2**0
                  CONTENTS, READONLY, DEBUGGING
 10 .debug_pubnames 00000035  0000000000000000  0000000000000000  00b02b08  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
 11 .debug_pubtypes 00000012  0000000000000000  0000000000000000  00b02b47  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
 12 .pdata        0000000c  0000000000000000  0000000000000000  00b02b63  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA
 13 .debug_line   000000d1  0000000000000000  0000000000000000  00b02b8d  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING
```

```
Patrick_Lehmann@Paebbels MINGW64 /c/Users/Patrick_Lehmann/Downloads
$ llvm-objdump --section-headers /c/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2/unisim/v08/unisim_VCOMP.o

C:/Tools/Precompiled/GHDL/4.0.0-dev/Vivado/2022.2/unisim/v08/unisim_VCOMP.o:    file format coff-x86-64

Sections:
Idx Name                                            Size     VMA              Type
  0 .text                                           00002f6f 0000000000000000 TEXT
  1 .data                                           00000000 0000000000000000 DATA
  2 .bss                                            000010cf 0000000000000000 BSS
  3 .xdata                                          00000010 0000000000000000 DATA
  4 .rdata                                          00821ebd 0000000000000000 DATA
  5 .rdata$.refptr.ieee__numeric_std__ELABORATED    00000008 0000000000000000 DATA
  6 .rdata$.refptr.ieee__std_logic_1164__ELABORATED 00000008 0000000000000000 DATA
  7 .debug_abbrev                                   00000028 0000000000000000 DATA, DEBUG
  8 .debug_info                                     0000003f 0000000000000000 DATA, DEBUG
  9 .debug_str                                      0000009b 0000000000000000 DATA, DEBUG
 10 .debug_pubnames                                 00000035 0000000000000000 DATA, DEBUG
 11 .debug_pubtypes                                 00000012 0000000000000000 DATA, DEBUG
 12 .pdata                                          0000000c 0000000000000000 DATA
 13 .debug_line                                     000000d1 0000000000000000 DATA, DEBUG
```

Somehow the section names like `//AAxHkl` are changing from analyze to analyze.

tgingold, 2023-05-07:
I need to better understand why these sections are created.
Maybe if the generated code is slightly different, the problem could be work-arounded.

tgingold, 2023-05-09:
I think I finally understand the issue.
It's an incompatibility between llvm and gnu ld.
For long section names and when the string table is large, llvm use a base64 encoding starting with '//'.
This is not supported by the gnu tools.  That's why you can see strange section names in the objdump output.

The workaround is to link with lld (the llvm linker).

Maybe we should also fix the issue in gnu tools (not too difficult), but it will take time until the change propagate to distributions.

skoehler, 2023-05-11:
I have generated a C file with 80.000 variable declarations. The names of the variables are longer than 128 characters.

The GNU assembler (mingw-w64-binutils 2.38-3) cannot generate the object file. The error message is "file too big". Ha ha. Given that it's possible to reproduce the issue without GHDL, it should be easy to report the bug to the binutils developers.

tgingold, 2023-05-12:
You are describing a different issue.  You get an assembler error but this issue is about a linker error.
Also, in this issue the object file is generated by llvm.

I have submitted a fix to binutils.

skoehler, 2023-05-12:
It's all related. The binutils' assembler can't write COFF files with base64 encoded section name indexes and binutils' ld, objdump, and friends can't read them.

Now I just wonder, whether it's an LLVM-specific extension or not. Will try Microsoft's compiler tonight. If that also creates base64 indexes, then we have a strong argument why binutils needs to be fixed.

tgingold, 2023-05-12:
Let us know, that would be interesting.

skoehler, 2023-05-12:
I was not able to make the microsoft compiler create object files that use the base64 string indexes. But the good news is that microsoft's `dumpbin.exe` (equivalent of objdump) decodes those base64 encoded indexes just fine. This is evidence that Microsoft supports base64 encoded indexes and that somewhere there should be some specification for that. Unfortunately, I didn't find the specification yet. `dumpbin.exe` calls the file format an `EXTENDED COFF OBJECT` while it calls files without base64 encoded names simply `COFF OBJECT`.

What I found so far is this: https://learn.microsoft.com/en-us/windows/win32/debug/pe-format
It says the following on Section names:
```
An 8-byte, null-padded UTF-8 encoded string. If the string is exactly 8 characters long, there is no terminating null. For longer names, this field contains a slash (/) that is followed by an ASCII representation of a decimal number that is an offset into the string table. Executable images do not use a string table and do not support section names longer than 8 characters. Long names in object files are truncated if they are emitted to an executable file.
```

No mention of section names that start with a double slash.

skoehler, 2023-05-14:
https://sourceware.org/bugzilla/show_bug.cgi?id=30444