ghdl/ghdl #2184: error: choice must be locally static expression

jamochl, 2022-08-30:
**Description**
A clear and concise description of what the issue is about.

This error shows up when using imported constant from another package

```
ALU.vhd:37:12:error: choice must be locally static expression
      when ALUopAdd =>
           ^
ALU.vhd:41:12:error: choice must be locally static expression
      when ALUopSub =>
           ^
```

**Expected behaviour**
What you expected to happen, and what is happening instead.

When constants are defined in a different package / file, eg.

```vhd :file: cpupackage.vhd
library ieee;
use ieee.std_logic_1164.ALL;
use ieee.numeric_std.all;

package cpupackage is
  constant ALUopAdd     : std_logic_vector := "000";
  constant ALUopSub     : std_logic_vector := "001";
end package cpupackage;
```

And that file is imported in another file

```vhd :file: alu.vhd
library ieee;
use ieee.std_logic_1164.ALL;
use ieee.numeric_std.all;

use work.cpupackage.ALL;

entity alu is
    Port ( aluOp     : in  std_logic_vector( 2 downto 0);
               O : out std_logic );
end alu;

architecture Behavioral of alu is
begin

ALUProcess:
process ( aluOp )
begin
   case aluOp is
      when ALUopAdd =>
          O <= '0';
      when ALUopSub =>
          O <= '1';
   end case;
end process ALUProcess;

end Behavioral;
```

I compile cpupackage package first, and then this ALU, and it should work, it works in modelsim. But in GHDL I get this error

```
ALU.vhd:37:12:error: choice must be locally static expression
      when ALUopAdd =>
           ^
ALU.vhd:41:12:error: choice must be locally static expression
      when ALUopSub =>
           ^
```

Tried it with `-frelaxed`, `--std=08`, and `-i *.vhd`, but nothing worked

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: cpupackage.vhd
library ieee;
use ieee.std_logic_1164.ALL;
use ieee.numeric_std.all;

package cpupackage is
  constant ALUopAdd     : std_logic_vector := "000";
  constant ALUopSub     : std_logic_vector := "001";
end package cpupackage;
```

```vhd :file: alu.vhd
library ieee;
use ieee.std_logic_1164.ALL;
use ieee.numeric_std.all;

use work.cpupackage.ALL;

entity alu is
    Port ( aluOp     : in  std_logic_vector( 2 downto 0);
               O : out std_logic );
end alu;

architecture Behavioral of alu is
begin

ALUProcess:
process ( aluOp )
begin
   case aluOp is
      when ALUopAdd =>
          O <= '0';
      when ALUopSub =>
          O <= '1';
   end case;
end process ALUProcess;

end Behavioral;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a cpupackage.vhd
ghdl -a alu.vhd
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: MacOS 21.6.0
- Origin:
  - [x] Package manager: `brew`
  - [x] Released binaries: `3.0.0-dev`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2022-08-31:
ghdl is correct here.  You need to constrain the constants.  Write:

```
package cpupackage is
  constant ALUopAdd     : std_logic_vector (2 downto 0):= "000";
  constant ALUopSub     : std_logic_vector (2 downto 0):= "001";
end package cpupackage;
```

jamochl, 2022-08-31:
That works for me, odd but ghdl does seem to be more correct. Thanks @tgingold

closed 2022-08-31 by jamochl