ghdl/ghdl #2185: GHDL Bug occurred

MJoergen, 2022-09-01:
**Description**
GHDL crashes when running simulation.

Output on terminal:
```
../../src/ieee2008/numeric_std-body.vhdl:1265:7:@0ms:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1265:7:@0ms:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1265:7:@0ms:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1265:7:@0ms:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:1873:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@10ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:1776:7:@10ns:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
avm_memory.vhd:60:13:@140ns:(report note): Writing 0xBFEE to 0x0 with burstcount 00
avm_memory.vhd:60:13:@150ns:(report note): Writing 0xBFEE to 0x0 with burstcount 00
```
followed by a "GHDL Bug occurred", see below.

**Expected behaviour**
No "GHDL Bug occurred".

**How to reproduce?**
[Avalon-mfj_arbit.zip](https://github.com/ghdl/ghdl/files/9468457/Avalon-mfj_arbit.zip)

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -i --std=08 --work=work avm_master_general.vhd avm_arbit.vhd avm_decrease.vhd avm_master.vhd avm_memory.vhd avm_pause.vhd burst_ctrl.vhd axi_avalon.vhd avalon_axi.vhd tb_avm_arbit.vhd
ghdl -m --std=08 -fexplicit tb_avm_arbit
ghdl -r --std=08 tb_avm_arbit --assert-level=error --wave=tb_avm_arbit.ghw --stop-time=50us
```

**Context**
Please, provide the following information:

- OS: "Linux Mint 21 Vanessa"
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `3f068da85` (Mon Aug 29 18:27:00 2022 +0200)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r665.g3f068da85) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-linux-gnu
/home/mike/git/MJoergen/Avalon/
Command line:
ghdl -r --std=08 tb_avm_arbit --assert-level=error --wave=tb_avm_arbit.ghw --stop-time=50us
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : SIGFPE
Call stack traceback locations:
0x7fbc484fe51e 0x7fbc47f61741 0x557b4817bd2d 0x557b4817c3c0 0x557b4817c975 0x557b4817ca0e 0x557b48116284 0x557b4828db27 0x557b4828dbfe 0x557b48538098 0x557b4846ef7c 0x557b4838326a 0x557b48538ab8 0x557b48115e0b 0x7fbc484e5d8e 0x7fbc484e5e3e 0x557b48114353 0xfffffffffffffffe
******************************************************************
```

MJoergen, 2022-09-01:
Running the same test using QuestaSim gives the error:

```
# ** Fatal: (vsim-3977) Integer divide (mod) by zero.
#    Time: 150 ns  Iteration: 1  Process: /tb_avm_arbit/i_avm_master_general1/i_avm_pause_master/p_cnt File: avm_pause.vhd
# Fatal error in Process p_cnt at avm_pause.vhd line 52
```

So that probably explains what triggered the GHDL bug.

tgingold, 2022-09-01:
Yup, on my machine I got:

ghdl_mcode:error: overflow detected
in process .***@***.******@***.***_pause(synthesis).p_cnt

I will try with a different linux distrib.

commit 3d7d9262 (2022-09-02): grt: add a SIGFPE handler for linux x86/64.  Fix #2185
closed 2022-09-02 by tgingold