ghdl/ghdl #2188: no file/linenumber information on runtime error with mcode64

tisvonkje, 2022-09-09:
**Description**
GHDL does not produce very elementary error messages, for example:
```
ghdl:error: overflow detected
ghdl:error: error during elaboration
```

**Expected behaviour**
I expect a more elaborated error message, something like:
```
ghdl:error: overflow detected
  from: Elaboration of design at Mux5Bits.vhdl:12
ghdl:error: error during elaboration
```

**How to reproduce?**
short error message is reproduced on macOS mcode64 ghdl

**Additional context**
Workaround would be to use llvm with backtrace library, but I could not get that working on MacOS

tisvonkje, 2022-09-09:
Managed to use workaround (with llvm and libbacktrace).

jobtijhuis, 2025-01-02:
I also ran into this problem. Makes the mcode version kind of useless if you don't get any good error messages.

Paebbels, 2025-01-02:
@tgingold can't we add `libbacktrace` also to mcode or was there a general problem with that library?

(It's long time ago when I looked into that part. I can't remember the results.)

tgingold, 2025-01-02:
Let me investigate why it is failing.  It shouldn't.

jobtijhuis, 2025-01-02:
I used the standalone Windows mcode build. Not sure if it should work in that version. I am not familiar with libbacktrace

alphanumericnonsense, 2025-05-09:
Any movement on this issue?  I never noticed it before (I think I was using gcc backend in the past), but now I'm using whatever is on packages.ubuntu.com for oracular (whatever version of Ubuntu WSL installed)

> $ ghdl --version
GHDL 4.1.0 (Ubuntu 4.1.0+dfsg-0ubuntu2.1) [Dunoon edition]
 Compiled with GNAT Version: 13.3.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

and I'm getting this mostly uninformative error.  Thanks for GHDL and all your hard work!

tgingold, 2025-05-10:
Yes, there have been changes.  Also, a reproducer would be helpful to precise which error message is not clear.