ghdl/ghdl #2190: Concatenation in port map with arrays doesn't work

HaarigerHarald, 2022-09-14:
**Description**
Concatenation in `port map` with arrays doesn't work.

**Expected behaviour**
It to work.

**How to reproduce?**

```vhd :file: test.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.all;

entity test_ent is
  port (
    vec   : in  std_logic_vector
  );
end test_ent;

library IEEE;
use IEEE.STD_LOGIC_1164.all;

entity test is
end test;

architecture rtl of test is
  constant a : std_logic_vector(2 downto 0) := "010";
  constant b : std_logic_vector(2 downto 0) := "101";
begin

  inst : entity work.test_ent
    port map(
      vec   => a(0) & b(1)
    );

end rtl;
```

```sh :image: ghdl/ghdl:ubuntu20-llvm-10
ghdl -a test.vhd
```

**Context**
Please, provide the following information:

- OS: CentOS7
- Origin:
  - [ ] Package manager:
  - [ ] Released binaries:
  - [x] Built from sources: `ff1ef30e8d370f89294e2d6e82fb1a15cdcd519c`

```
translate_name: cannot handle IIR_KIND_CONCATENATION_OPERATOR (ghdl_test.vhd:24:21)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: Community 2021 (20210519-103)
Target: x86_64-pc-linux-gnu
/home/behuber/
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ -c -o ghdl_test.o ghdl_test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x4f8e3a 0x6558aa 0x64cc17 0x64ccfb 0x64d140 0x64d63e 0x64e32e 0x6e0233 0x6dbfe3 0x6dc718 0x699c6d 0x68d46a 0x6e7244 0x6e86f8 0x6e98b1 0x40b5f9 0x7f0217f91553 0x40a38b
******************************************************************
```

tgingold, 2022-09-14:
I cannot reproduce the issue, so probably it has already been fixed.

HaarigerHarald, 2022-09-14:
Strange, I used the most recent commit and now also tested with mcode backend instead of llvm.

```vhd :file: test.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.all;

entity test_ent is
  port (
    vec   : in  std_logic_vector
  );
end test_ent;

architecture rtl of test_ent is
begin
end rtl;

library IEEE;
use IEEE.STD_LOGIC_1164.all;

entity test is
end test;

architecture rtl of test is
  constant a : std_logic_vector(2 downto 0) := "010";
  constant b : std_logic_vector(2 downto 0) := "101";
begin

  inst : entity work.test_ent
    port map(
      vec   => a(0) & b(1)
    );

end rtl;
```

```
ghdl -a --std=08 test.vhd
ghdl -r --std=08 test
```

Produces:

```
translate_name: cannot handle IIR_KIND_CONCATENATION_OPERATOR (ghdl_test.vhd:27:21)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r689.gff1ef30e8) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: i686-w64-mingw32
\\wsl$\CentOS7\home\behuber\
Command line:
C:\bin\ghdl.exe -r --std=08 test
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Load address: 0xe70000
Call stack traceback locations:
0x116e557 0x116e96d 0x116f55d 0x11ea583 0x11e69c0 0x11e7199 0x11ad15f 0x11a2c7a 0x11b1902 0x11abda9 0x11eec57 0x113cfe9 0x106ecb0 0x11f17e9 0xe72e20 0xe7138c 0x76defa27 0x77997a9c 0x77997a6c
******************************************************************
```

tgingold, 2022-09-14:
Ah ok, I was testing with --std=08.

n8tlarsen, 2024-01-11:
I'm also having this issue. Has any progress been made?

tgingold, 2024-01-11:
Which version of ghdl are you using ? I cannot reproduce the crash.

n8tlarsen, 2024-01-11:
Thanks for the quick response. I didn't realize there was a new release; I was using version 2.0.0. Version 3.0.0 does not exhibit the issue.

HaarigerHarald, 2024-01-11:
I think this indeed has been resolved in 3.0.0, as I also cannot reproduce it with this version anymore.

closed 2024-01-11 by HaarigerHarald

n8tlarsen, 2024-03-01:
I found this issue pop up again in GHDL version 3.0.0 with the llvm backend. When the port definition contains an unconstrained array, the same crash occurs. All concatenated signals are constrained.

tgingold, 2024-03-02:
It is exactly the same issue with the same reproducer ?  This test is in the regression testsuite, so it doesn't fail with the current ghdl,

n8tlarsen, 2024-07-08:
I just installed version 4.1.0 and this issue still exists in my case. Here's the entity:

``` vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity skid_buffer_in is
    port (
        aclk    : in  std_logic;
        aresetn : in  std_logic;
        -- registered
        r_valid : in  std_logic;
        r_ready : out std_logic;
        r_data  : in  std_logic_vector;
        -- combinational
        c_valid : out std_logic;
        c_ready : in  std_logic;
        c_data  : out std_logic_vector
    );
end entity skid_buffer_in;

architecture rtl of skid_buffer_in is
    constant data_width : natural := r_data'length;
    signal b_data  : std_logic_vector(r_data'range);
    signal b_valid : std_logic; 
begin

    skid : process(aclk, aresetn)
    begin
        if (aresetn = '0') then
            r_ready <= '0';
            b_data  <= (others => '0');
            b_valid <= '0';
        elsif rising_edge(aclk) then
            if c_ready then
                b_valid <= '0';
            end if;
            if (r_valid and r_ready) = '1' then
                if c_ready = '0' then
                    b_data <= r_data;
                end if;
                b_valid <= not c_ready;
                r_ready <= c_ready;
            else
                r_ready <= c_ready or not b_valid;
            end if;
        end if;
    end process skid;

    with to_bit(b_valid) select c_data <= 
        r_data when '0',
        b_data when '1';
    with to_bit(b_valid) select c_valid <= 
        r_valid and r_ready when '0',
        '1' when '1';
    
end architecture rtl;
```
and an example instantiating entity (please excuse syntax errors as this is an excerpt from larger file):
``` vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity inst is
    port (
        aclk    : in  std_logic;
        aresetn : in  std_logic;
        wvalid  : in  std_logic;
        wready  : out std_logic;
        wdata   : in  std_logic_vector(31 downto 0);
        wstrb   : in  std_logic_vector( 3 downto 0)
    );
end entity inst;

architecture struct of inst is
    constant data_bytes : natural := 4;
    signal bi_wvalid, bi_wready   : std_logic;
    signal bi_wdata : std_logic_vector(data_bytes*8-1 downto 0);
    signal bi_wstrb : std_logic_vector(data_bytes-1   downto 0);
begin

    w_skid : entity work.skid_buffer_in
    port map(
        aclk    => aclk,
        aresetn => aresetn,
        r_valid => wvalid,
        r_ready => wready,
        r_data  => wstrb & wdata,
        c_valid => bi_wvalid,
        c_ready => bi_wready,
        c_data(wdata'length+wstrb'high downto wdata'length) => bi_wstrb,
        c_data(wdata'high downto 0) => bi_wdata
    );

end architecture struct;
```
Surprisingly the following subsitution does not exhibit the internal error, but instead the simulation error: "slice direction doesn't match index direction"
``` vhdl
r_data(wdata'length+wstrb'high downto 0)  => wstrb & wdata,
```

n8tlarsen, 2024-07-09:
I tested with mcode and llvm backends. The mcode backend throws the error when starting simulation while llvm throws the error at compile time.

tgingold, 2024-07-10:
You should really create a new issue in that case.  This issue is already fixed and closed.

commit 430bfd49 (2024-01-12): testsuite/gna: add a reproducer for #2190

Close #2190