ghdl/ghdl #2191: synth mem2d01: failed during build

SiliconWizard, 2022-09-21:
ghdl-mcode currently fails to build.

The error is:
` synth mem2d01: failed`
from ../suite_driver.sh, line 44
`make: *** [Makefile:231 : test.mcode] Erreur 1`

tgingold, 2022-09-21:
Humm, what is the version of ghdl you are using ?

SiliconWizard, 2022-09-22:
Just trying to build from the latest git source code. Not from a release. I realize this may not always build. :)
Although I've had good success with that until recently with this test error. Just pulled the latest a couple minutes ago and still get the same failed test. It's a relatively recent "build" issue, unfortunately I can't tell from what exact revision this problem appeared.

tgingold, 2022-09-22:
Which commit are you using ?

tgingold, 2022-09-22:
Your repo should be old, because of:

commit 119034986fb631d2e8baa8e90aa30febe5b95b55
Author: Tristan Gingold ***@***.***>
Date:   Sun Aug 14 04:43:51 2022 +0200

    testsuite/synth: rename mem2d01 to memdp01

SiliconWizard, 2022-09-22:
I was using the last commit, but I forgot to clean and relauch the configuration. So it was using the latest source code but a configuration older than commit https://github.com/ghdl/ghdl/commit/119034986fb631d2e8baa8e90aa30febe5b95b55

Sorry about the noise and thanks for the pointer though!

closed 2022-09-23 by tgingold