ghdl/ghdl #2193: "overflow detected" in file reading

thesps, 2022-09-22:
Thanks for the nice project! I'm trying to port some testbench simulations I have over to GHDL, and in doing so I hit this issue.

**Description**
Trying to read a file in a testbench, I hit an "overflow detected" error from GHDL when running the simulating, when slicing a `string` from a `line`.

**Expected behaviour**
The line string slicing shouldn't cause an overflow. Indeed this simulates okay with Modelsim.

**How to reproduce?**
Here's a minimal working example. It doesn't use file IO, here I just assigned `line` from a `string`, but the same "overflow detected" is hit when reading the `line` from a file instead. This works fine if commenting out the `read(l, s)` line

```vhd :file: demo.vhd
library ieee;
use ieee.std_logic_1164.all;
use std.textio.all;

entity demo is
end demo;

architecture behavioral of demo is
    signal clk  : std_logic := '0';
    signal i    : integer   := 0;
begin

  clk <= not clk after 5 ns;

  process(clk)
    variable l : line;
    variable s : string(1 to 4) := "    ";
  begin
    if rising_edge(clk) then
      if i = 1 then
        std.env.finish;
      end if;
      l := new string(1 to 13);
      l.all := "hereisastring";
      report l.all; -- fine
      report l.all(1 to 4); -- fine
      read(l, s); -- fine if commented out
      report s;
      report l.all; -- fine
      report l.all(1 to 4); -- overflow detected
      i <= i+1;
    end if;
  end process;

end behavioral;

```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -c --std=08 -fsynopsys demo.vhd
ghdl -r --std=08 -fsynopsys
```

Output from ghdl run:
```
demo.vhd:25:7:@5ns:(report note): hereisastring
demo.vhd:26:7:@5ns:(report note): here
demo.vhd:28:7:@5ns:(report note): here
demo.vhd:29:7:@5ns:(report note): isastring
ghdl:error: overflow detected
in process .demo(behavioral).P1
  from: process work.demo(behavioral).P1 at demo.vhd:30
ghdl:error: simulation failed
```

For reference, expected output from Modelsim
```
#  run -all
# ** Note: hereisastring
#    Time: 5 ns  Iteration: 0  Instance: /demo
# ** Note: here
#    Time: 5 ns  Iteration: 0  Instance: /demo
# ** Note: here
#    Time: 5 ns  Iteration: 0  Instance: /demo
# ** Note: isastring
#    Time: 5 ns  Iteration: 0  Instance: /demo
# ** Note: isas
#    Time: 5 ns  Iteration: 0  Instance: /demo
```
**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin:
```
$ ghdl --version
GHDL 2.0.0 (tarball) [Dunoon edition]
```

tgingold, 2022-09-23:
I fear it's because you rely on undefined behaviour: the bounds of the line parameter are not defined after the call to the read procedure.
Try 'report l.all(5 to 13);`

thesps, 2022-09-23:
Okay thanks a lot, I didn't realise it was undefined behaviour! The suggested fix works 👍

commit 91a072a3 (2022-09-30): testsuite/gna: add a test for #2193

close #2193
closed 2022-09-30 by tgingold