ghdl/ghdl #2194: Modeling changes for https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/275

JimLewis, 2022-09-23:
**Description**
I am modeling behavior for https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/275 using generate.   It runs, but I do not get any of the log messages.   

**Expected behaviour**
I expect assert notes of the following form:
# ** Note: V = U1U1U1U1U1UU
# ** Note: V = 01010101010U

No output was produced

**How to reproduce?**

```vhd :file: TestLongestStaticPrefixFuncGen .vhd
library IEEE ;
use ieee.std_logic_1164.all ; 

entity TestLongestStaticPrefixFuncGen is
end entity TestLongestStaticPrefixFuncGen ;
architecture t1 of TestLongestStaticPrefixFuncGen is 
  signal v : std_logic_vector(11 downto 0) := (others => '-') ; 
  
  function func(i : integer) return integer is 
  begin
    return 2 * i ; 
  end function func ; 
  
  signal A : integer ; 
  
begin

  process 
  begin
    wait for 1 ns ; 
    report  "V = " & to_string(V) ; 
    A <= 5 ; 
    wait for 1 ns ; 
    report  "V = " & to_string(V) ; 
    wait ;
  end process ; 

  gen1 : for i in 0 to 5 generate
    process (v)
    begin
      if i /= 0 then 
        v(func(i)) <= '1' ;
      end if; 
    end process ; 
    
    process (v, A)
    begin
      if A > 2 then 
        v(func(i) + 1) <= '0' ; 
      end if ; 
    end process ; 
  end generate gen1 ; 

end architecture t1 ; 

```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm_DpRam --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/osvvm_dpram/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev ../../SandBox/TestLongestStaticPrefixFuncGen.vhd
ghdl --elab-run --std=08 --syn-binding --work=osvvm_DpRam --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/osvvm_dpram/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev TestLongestStaticPrefixFuncGen
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

JimLewis, 2022-09-23:
I should also point out that this is a "toy" example and is not anything I need at this point in time.   I am using this example to explore what tools already can do with respect to `for loops` by modeling it with `for generate`

diogratia, 2022-09-23:
For the last nightly build commit a259712cc 3 weeks ago
testsuite/synth: improve test #1460	a259712cc	Tristan Gingold <tgingold@free.fr>	Sep 2, 2022 at 12:31 PM (time and date GMT-11)

```none
%: setenv GHDL ~/Desktop/ghdl/2.0.0.r670.ga259712cc/ghdl-macos-12-mcode/bin/ghdl
%: $GHDL --version
GHDL 3.0.0-dev (2.0.0.r670.ga259712cc) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
%: $GHDL -a --std=08 testlongeststaticprefixfuncgen.vhdl
%: $GHDL --elab-run --std=08 testlongeststaticprefixfuncgen
testlongeststaticprefixfuncgen.vhdl:21:5:@1ns:(report note): V = -1-1-1-1-1--
testlongeststaticprefixfuncgen.vhdl:24:5:@2ns:(report note): V = 01010101010-
%: 
```

You may need to provide more information.

JimLewis, 2022-09-24:
Is that printed on STD OUTPUT or STD_ERROR?

JimLewis, 2022-09-24:
Thanks.   I found the issue.   It was TCL catch + exec catches all simulator output.

closed 2022-09-24 by JimLewis

tgingold, 2022-10-11:
Which version of ghdl are you using ?  I get:

```
$ ./testlongeststaticprefixfuncgen 
***@***.***:(report note): V = -1-1-1-1-1--
***@***.***:(report note): V = 01010101010-
```

JimLewis, 2022-10-11:
I closed this because it was an OSVVM scripting bug that was suppressing the output when running in GHDL.   So I mistakenly thought it was GHDL when it was OSVVM scripts.   I was expecting catch + exec to only catch the errors (like catch does when exec is not used), but instead it was also catching all output.  

This issue is fixed on the OSVVM dev branch.